ros/ros_comm #746: Build servers fail

eric-wieser, 2016-02-21:
See [this page](http://build.ros.org/job/Jpr__ros_comm__ubuntu_trusty_amd64/20/testReport/junit/test.test_roswtf_command_line_offline/TestRoswtfOffline/test_offline/)

The tests produces a warning about a missing `rosinstall`, which causes the test to fail. Is `rosinstall` deliberately omitted?

dirk-thomas, 2016-02-22:
The message about rosinstall not being available is only a warning and not the reason why this test is failing.

The problem is related to the new build farm, how catkin sets environment variables (even when not being necessary) and roswtf testing if anything is set which isn't necessary. It will be fixed once the patch merged in ros/catkin#777 is available from Debian packages and therefore being used by the PR jobs for this repo.

eric-wieser, 2016-02-22:
The test looks for the string "No warnings or errors" to determine if a pass occurs. So wouldn't the initial print of this warning cause a failure? Or did I not read the test carefully enough

dirk-thomas, 2016-02-22:
That is true. Once the error is solved by the catkin patch the test still needs to be updated to not require `rosinstall`.

eric-wieser, 2016-02-22:
Can this test be removed until it passes, or marked as an expected failure by jenkins somehow? It's unhelpful to have CI run on all prs when it never passed to begin with

dirk-thomas, 2016-02-22:
The release of catkin will happen in the next few days and then I will also make sure that the warning about `rosinstall` is removed. Until now please just ignore the one failing test. If everything else is passing your PR is considered "green".

eric-wieser, 2016-02-23:
Related issue - the build servers now email me when _anyone_'s PR fails. How do I unsubscribe?

dirk-thomas, 2016-02-23:
The build server sends email to all contributors which have committed to the branch since it was last time stable.

I have created #752 even if the problem should be addressed very soon.

eric-wieser, 2016-02-23:
>  have committed to the branch

Which branch? Aren't we all committing to our own independent feature branches? It strikes me as broken behaviour that if my PR fails, then I get notified for every other person who makes their own failing PR, up until someone succeeds

dirk-thomas, 2016-02-24:
@eric-wieser We are not using any custom notification configuration. All maintainers of the packages are being notified as well as all contributors as determined by the Jenkins plugin.

dirk-thomas, 2016-02-24:
For the rosinstall warning I have created #756.

PR #752 (2016-02-24): temporary disable failing roswtf test
Fixes #746.
closed 2016-02-24 by dirk-thomas