ros/ros_comm #748: ROS_LOG_THROTTLE not working when ros::time goes backward

hfarazi, 2016-02-22:
Sometimes I need to play a bag file in a loop (with -l) in those times any ROS_LOG_THROTTLE is not working.

It seems that the problem is related to the definition of last_hit in ROS_LOG_THROTTLE:
static double last_hit = 0.0;

My suggestion is reset the last_hits when an empty message arrived in /reset_time topic.

hfarazi, 2016-04-25:
it can be easily fixed by using:

#define ROS_LOG_THROTTLE2(rate, level, name, ...) \
  do \
  { \
    ROSCONSOLE_DEFINE_LOCATION(true, level, name); \
    static double last_hit = 0.0; \
    ::ros::WallTime now = ::ros::WallTime::now(); \
    if (ROS_UNLIKELY(__rosconsole_define_location__enabled) && ROS_UNLIKELY(last_hit + rate <= now.toSec())) \
    { \
      last_hit = now.toSec(); \
      ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
    } \
  } while(0)

dirk-thomas, 2016-04-25:
Please consider to create a pull request for a proposed code change. That way the patch can be compiled and tested automatically.

dirk-thomas, 2018-05-21:
This issue was moved to ros/rosconsole#6

closed 2018-05-21 by dirk-thomas