ros/ros_comm #753: Add return value to void node_handle.param

NikolasE, 2016-02-23:
There is no way to see if the default value for a loaded parameter was used or if it was read from the parameter server. The function currently does not return anything so it would be simple to improve it like this:

  template<typename T>
  **bool** param(const std::string& param_name, T& param_val, const T& default_val) const
  {
    if (hasParam(param_name))
    {
      if (getParam(param_name, param_val))
      {
        return **true**;
      }
    }
    param_val = default_val;
    return **false**;
}

What do you think? Current code would just ignore the return value so that it should not create a problem for anyone.

dirk-thomas, 2016-02-23:
The proposed change would change ABI and therefore not be consider for already released distros (Indigo, Jade). But it could easily be implemented for Kinetic since it doesn't require any change in user code. Would you be interested to provide a pull request including some tests once there is a `kinetic-devel` branch?

There is a simple way to achieve the desired behavior with the existing API with the only downside that it requires more then one line. You can call `bool getParam(key, & value)` which will set the parameter value if it is available on the parameter server and if the return value is false you have to set the variable to a default value manually.

NikolasE, 2016-03-02:
Hey! I'm going to write the pull request. Maybe we could also (sooner) add a
`bool get_setter("my_param", param_value, default_value)`
that would return hasParam("my_param") and would either read the value from the server or use the default_value and write it to the server.

dirk-thomas, 2016-03-02:
I am not sure that it is a good idea that a function caled `get_*` actually writes any data to the server. But it could certainly provide the two separate steps in a single function.

dirk-thomas, 2016-03-17:
Please see #775 for the proposed change in Kinetic.

closed 2016-03-18 by dirk-thomas

NikolasE, 2016-03-19:
Look perfect. Thanks.