ros/ros_comm #768: View with reduce_overlap=true can segfault

cpwais, 2016-03-16:
I may be using `View`'s  `reduce_overlap` feature incorrectly, but nevertheless I discover that it segfaults on a simple example such as the one below:

```
  using namespace rosbag;
  ros::Time::init();
  Bag bagw;
  bagw.open("/tmp/bug_demo.bag", rosbag::BagMode::Write);
  std_msgs::String m;
  m.data = "test";
  bagw.write("test_topic", ros::Time::now(), m);
  bagw.close();

  Bag bagr;
  bagr.open("/tmp/bug_demo.bag", rosbag::BagMode::Read);
  View v(/* reduce_overlap */ true); // false --> no segfault
  v.addQuery(bagr);
  BOOST_FOREACH(rosbag::MessageInstance const m, v) {
    std::cout << "message! " << &m << std::endl;
  }
```

This program prints "message!" *twice* (expected once) and then segfaults.

I'm compiling my program for `c++11` and using my own build of ROS indigo via `catkin_make_isolated  --install --install-space /opt/ros/indigo -DCMAKE_CXX_COMPILER=/usr/bin/clang++ -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-O0 -g3 -std=c++11 -stdlib=libc++"`.  

The segfault occurs in `rosbag::View::iterator::increment()` (from valgrind):
```
==21188==    by 0x304A47F2: __tree_next<std::__1::__tree_node_base<void *> *> (__tree:68)
==21188==    by 0x304A47F2: operator++ (__tree:747)
==21188==    by 0x304A47F2: operator++ (__tree:751)
==21188==    by 0x304A47F2: rosbag::View::iterator::increment() (view.cpp:128)
==21188==    by 0x2FADE7E4: void boost::iterator_core_access::increment<rosbag::View::iterator>(rosbag::View::iterator&) (iterator_facade.hpp:520)
```

I tried groking the `::increment()` source but I don't understand it enough to suggest a fix.  Also I'm a little curious if there might be a `libc++`-related issue at play here... has this feature been tested with `libstdc++`?

More info about setup:
```
# clang++ --version
Ubuntu clang version 3.4-1ubuntu3 (tags/RELEASE_34/final) (based on LLVM 3.4)
Target: x86_64-pc-linux-gnu
Thread model: posix
```

```
# strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep LIBCXX
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_DEBUG_MESSAGE_LENGTH
```

dirk-thomas, 2016-03-17:
WIth a reproducible example it is hard to tell if this is a generic problem or not. Please share a minimal bagfile which exposes the problem. It would be even better if you could create a unit test for the problem (which currently fails). That way we can reliably try it on different platforms and also confirm that a potential patch resolved the issue.

cpwais, 2016-03-17:
Note sure what you mean by: "WIth a reproducible example it is hard to tell if this is a generic problem or not"

did you mean to write "without" ?   But did you see the code snippet I posted?  did that work for you? when compiled using the flags I cited?

wjwwood, 2016-03-17:
@cpwais I think he meant "without a reproducible example". I think he is also referring to the desire to have a small bag file which when used in combination with the example you gave that fails. I don't know if he tried it with some other arbitrary bag and didn't reproduce it with your example or not. I haven't tried myself.

We do appreciate the detailed report.

cpwais, 2016-03-17:
ok cool thanks!  I played that code snippet in my build and got the segfault.  I believe I included all relevant build deets but LMK if you're having trouble reproducing.  (I'm actually not sure if my case is just an unexpected case for `reduce_overlap` ... maybe `reduce_overlap` is only for views that have multiple bags and should turn itself off if there's only one bag).

The error seems to be in `View()`'s `reduce_overlap` mode.  The `rosbag::View::iterator` never appears to equal `View::end()` and then `::increment()` eventually segfaults.  I could not make enough sense of what `increment()` was doing with its internal iterators to think of a fix.

dirk-thomas, 2016-03-17:
I have created a unit test with your snippet (https://github.com/ros/ros_comm/compare/test_view_reduce_overlay?expand=1) and it passes for me without a problem. Please run this test locally and comment about the result.

cpwais, 2016-03-17:
Looks like this bug is probably c++11-related.  AFAICT that test is not compiling for C++11 and/or not using a ROS install compiled against `libc++` (`-DCMAKE_CXX_FLAGS="-O0 -g3 -std=c++11 -stdlib=libc++"` as noted above).

dirk-thomas, 2016-03-17:
Also with `--std=c++11` it passes for me.

I can't compile against `libc++` on Trusty out-of-the box.

Please try to run the test locally and comment about the actual result you are seeing with it.

dhood, 2016-03-22:
@cpwais, I've been able to reproduce your error by having only some of my ROS packages compiled against `libc++`. This makes me think this might be caused by an ABI mis-match, similar to the (non-ROS) situation [here](http://stackoverflow.com/questions/16384659/c-std-c11-stdlib-libc-with-boost-thread-gives-segmentation-fault-11-on). 

Is there a possibility that some of the packages on your install path are built with gcc? or have you been able to build everything from scratch against `libc++`? Personally I wasn't able to do a complete build of the relevant ROS packages from scratch against `libc++`, because rosbag would need log4cxx compiled against `libc++`, etc.

@dirk-thomas the test output is as follows if ROS packages are built in release (if ROS packages are built in debug it doesn't segfault)
```
Scanning dependencies of target _run_tests_test_rosbag_storage_gtest_view_reduce_overlay
-- run_tests.py: execute commands
  /home/dhood/src/ros_comm/test/test_rosbag_storage/build/devel/lib/test_rosbag_storage/view_reduce_overlay --gtest_output=xml:/home/dhood/src/ros_comm/test/test_rosbag_storage/build/test_results/test_rosbag_storage/gtest-view_reduce_overlay.xml
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from rosbag_storage
[ RUN      ] rosbag_storage.view_reduce_overlay
message! without boost 0x7fffcde4f3f0
message! without boost 0x7fffcde4f3f0
Segmentation fault (core dumped)
-- run_tests.py: verify result "/home/dhood/src/ros_comm/test/test_rosbag_storage/build/test_results/test_rosbag_storage/gtest-view_reduce_overlay.xml"
Cannot find results, writing failure results to '/home/dhood/src/ros_comm/test/test_rosbag_storage/build/test_results/test_rosbag_storage/MISSING-gtest-view_reduce_overlay.xml'
```
Note that getting the test to build with `-DCMAKE_CXX_FLAGS="-std=c++11 -stdlib=libc++" -DCMAKE_CXX_COMPILER=/usr/bin/clang++` required updating my gtest from 1.6.0 to 1.7.0.

cpwais, 2016-03-22:
Sorry for the delay and thanks @dhood for the extra details.

I had also built boost against `libc++`.  For more context here's part of the Dockerfile for my build image:

```
FROM ubuntu:trusty-20160226

RUN apt-get update
RUN apt-get -y install git build-essential ccache clang libc++-dev cmake wget vim python-pip
ENV CXX clang++

RUN apt-get install -y python-dev liblz4-dev libbz2-dev zlib1g-dev
WORKDIR /tmp
RUN wget -O boost.tar.gz https://sourceforge.net/projects/boost/files/boost/1.56.0/boost_1_56_0.tar.gz/download
RUN tar xvzf boost.tar.gz
WORKDIR /tmp/boost_1_56_0
RUN ./bootstrap.sh --with-toolset=clang --prefix=/usr/local
RUN ./b2 clean
RUN ./b2 toolset=clang cxxflags="-std=c++11 -stdlib=libc++ -O3" linkflags="-stdlib=libc++" --prefix=/usr/local -j 10 define=BOOST_SYSTEM_NO_DEPRECATED stage release
RUN ./b2 install toolset=clang cxxflags="-std=c++11 -stdlib=libc++ -O3" linkflags="-stdlib=libc++" --prefix=/usr/local

# ROS (from source for c++11 support)

## Catkin
WORKDIR /tmp
RUN git clone --depth 1 --single-branch -b indigo-devel https://github.com/ros/catkin.git
RUN apt-get -y install python-pip
RUN pip install nose empy catkin_pkg
WORKDIR /tmp/catkin/build
RUN cmake ..
RUN make -j8 && make install

## console_bridge
WORKDIR /tmp
RUN git clone --depth 1 git://github.com/ros/console_bridge.git
WORKDIR /tmp/console_bridge
RUN cmake -DCMAKE_CXX_FLAGS="-O3 -std=c++11 -stdlib=libc++" . && make && make install

## ROS Core
WORKDIR /tmp
RUN apt-get install -y libtinyxml-dev
RUN pip install rosinstall_generator wstool
WORKDIR /tmp/ws
RUN catkin_init_workspace
RUN rosinstall_generator rosconsole rosconsole_bridge ros_comm diagnostic_msgs --rosdistro indigo --deps --wet-only > indigo-ros_comm-wet.rosinstall
RUN wstool init -j10 src indigo-ros_comm-wet.rosinstall
  # NB: if the above fails, use $ wstool update -j 10 -t src
RUN ./src/catkin/bin/catkin_make_isolated --install --install-space /opt/ros/indigo -DCMAKE_CXX_COMPILER=`which clang++` -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-O3 -g3 -std=c++11 -stdlib=libc++"

```

I did not have a problem building `log4cxx` for some reason ...   I am definitely linked to `libc++` and not `libstdc++`.  The segfault appears to happen in the STL headers tho (see `__tree_next` in the traceback).  Note this segfault appears *after* the iterator should have been equal to `View::end()`.  So I think there's a subtle programming bug here ... and that perhaps `View` with `reduce_overlap` should do nothing special if there's only one bag.

More deets:
```
# dpkg -s libc++-dev | grep Version
Version: 1.0~svn199600-1

# ldd /opt/ros/indigo/lib/librosbag*
/opt/ros/indigo/lib/librosbag.so:
	librosbag_storage.so => /opt/ros/indigo/lib/librosbag_storage.so (0x00007f9a3d470000)
...
	libc++.so.1 => /usr/lib/x86_64-linux-gnu/libc++.so.1 (0x00007f9a3a74b000)
/opt/ros/indigo/lib/librosbag_storage.so:
	libroscpp_serialization.so => /opt/ros/indigo/lib/libroscpp_serialization.so (0x00007f6ca226e000)
	librostime.so => /opt/ros/indigo/lib/librostime.so (0x00007f6ca2043000)
	libc++.so.1 => /usr/lib/x86_64-linux-gnu/libc++.so.1 (0x00007f6ca0556000)
...
```

dhood, 2016-03-22:
Thanks for the info, I'll see if I can have another look. Did you ever notice that the problem went away for you in debug, like happens for me?

dhood, 2016-03-24:
@cpwais I followed your dockerfile commands and was able to reproduce your error again with everything built against `libc++` as you said, thanks. 

I'm just getting the segfault if the rosbag_storage package is built with optimisation -O2 or -O3. From your dockerfile it also looks like you're using -O3, right? Does changing the optimisation fix it for you?

cpwais, 2016-03-25:
hmm, you're claiming there's no crash if no optimization?  I tried with `-O0 -g3` to debug (in the `catkin_make_isolated ` step but not boost I don't think) and I was still seeing the crash... I also want optimization

dhood, 2016-03-26:
that is what I saw on my machine (new trusty install), yeah, but I also see now for myself that the problem persists in your docker image. will try some variations to see what might be causing the difference, but open to suggestions if anyone's got them

dhood, 2016-03-29:
@cpwais,
Still not sure what's causing this but here's a workaround:

Dockerfile:
```
RUN ./b2 install toolset=clang cxxflags="-std=c++11 -stdlib=libc++ -O3" linkflags="-stdlib=libc++" --prefix=/usr/local

# ROS (from source for c++11 support)

## Catkin
WORKDIR /tmp
RUN git clone --depth 1 --single-branch -b indigo-devel https://github.com/ros/catkin.git
RUN apt-get -y install python-pip
RUN pip install nose empy catkin_pkg
WORKDIR /tmp/catkin/build
RUN cmake ..
RUN make -j8 && make install

## console_bridge
WORKDIR /tmp
RUN git clone --depth 1 git://github.com/ros/console_bridge.git
WORKDIR /tmp/console_bridge
RUN cmake -DCMAKE_CXX_FLAGS="-O3 -std=c++11 -stdlib=libc++" . && make && make install

## ROS Core
WORKDIR /tmp
RUN apt-get install -y libtinyxml-dev
RUN pip install rosinstall_generator wstool
WORKDIR /tmp/ws
RUN catkin_init_workspace
RUN rosinstall_generator rosconsole rosconsole_bridge ros_comm diagnostic_msgs --rosdistro indigo --deps --wet-only > indigo-ros_comm-wet.rosinstall
RUN wstool init -j10 src indigo-ros_comm-wet.rosinstall
  # NB: if the above fails, use $ wstool update -j 10 -t src
RUN ./src/catkin/bin/catkin_make_isolated --install --install-space /opt/ros/indigo -DCMAKE_CXX_COMPILER=`which clang++` -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-O3 -g3 -std=c++11 -stdlib=libc++"

## googletest
WORKDIR /tmp
RUN wget -O gtest.tar.gz https://github.com/google/googletest/archive/release-1.7.0.tar.gz
RUN tar xf gtest.tar.gz
WORKDIR /tmp/googletest-release-1.7.0
RUN cmake .
RUN make

## Workaround for rosbag_storage
RUN git clone --single-branch -b test_view_reduce_overlay https://github.com/ros/ros_comm.git /tmp/src/ros_comm
```

Building the branch with the tests in debug and -O0, your example code works as expected:

```
cd /tmp/src/ros_comm/tools/rosbag_storage
source /opt/ros/indigo/setup.bash 
catkin_make_isolated --source . --install --install-space /opt/ros/indigo -DCMAKE_CXX_COMPILER=`which clang++` -DCMAKE_BUILD_TYPE=Debug -DCMAKE_CXX_FLAGS="-O0 -g3 -std=c++11 -stdlib=libc++"
cd /tmp/src/ros_comm/test/test_rosbag_storage
mkdir build && cd build
cmake .. -DCMAKE_CXX_COMPILER=`which clang++` -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-O0 -g3 -std=c++11 -stdlib=libc++" -D_CATKIN_GTEST_INCLUDE=/tmp/googletest-release-1.7.0/include/gtest/gtest.h -D_CATKIN_GTEST_SRC=/tmp/googletest-release-1.7.0/src/gtest.c
make run_tests
```

Still segfaulting in Release, however.  Perhaps this can be of some use to you until the problem is resolved.

closed 2025-05-31 by sloretz