ros/ros_comm #769: rosbag python API deserializes to non-identity-equal types

eric-wieser, 2016-03-16:
All the types are created on the fly do not match the types importable, ie:

    bag = rosbag.Bag(...)
    msg = next(iter(bag))
    isinstance(msg, PoseStamped)

Will always return `False` on a bag containing stamped poses

dirk-thomas, 2016-03-17:
Currently the message type is being generated on-the-fly (https://github.com/ros/ros_comm/blob/indigo-devel/tools/rosbag/src/rosbag/bag.py#L1562-L1570) in order to handle arbitrary message types (which might not even be available on the system.

It would be possible to look at `info.datatype` which can be parsed by `genmsg.package_resource_name` and try to import the message definition. If that is available (and has the same md5sum) it could be used instead of the dynamically generated type.

I will likely not have the bandwidth to work on this but would be happy to review a merge a pull request.

eric-wieser, 2016-03-17:
Is making the change happen upstream in `genpy.dynamic.generate_dynamic` sensible?

dirk-thomas, 2016-03-17:
I would say no since `generate_dynamic` should still do what its name implies: generate a dynamic type.

eric-wieser, 2016-03-17:
I'd argue that you never want to have two identical but not identity-equal types floating around the system

eric-wieser, 2016-03-17:
Actually, I'd say this is required to give sane rosbag behaviour. If I have a custom message type `MyWrappedImage` containing an `sensor_msgs/Image`, and `MyWrappedImage` isn't installed on my system, I'd expect it to dynamically generate a class that uses the static `sensor_msgs/Image`. Right now, that gets dynamically generated too.

dirk-thomas, 2016-03-17:
The name of the function in `genpy` implies a very specific semantic behavior. Using existing message definitions when available should happen outside of that function.

This conditional logic can be put anywhere (maybe even in `genpy`) and then be used wherever useful.

eric-wieser, 2016-03-17:
Sounds like `generate_dynamic` needs to acquire a `use_existing = False` default argument then

dirk-thomas, 2016-03-17:
As argued above I don't think it is in the scope of a function called `generate_dynamic` to return non-dynamic message types.

A new function `get_message_type` should be introduced which either returns an existing message class or the result of `generate_dynamic` otherwise.

eric-wieser, 2016-03-17:
Seems reasonable

mikepurvis, 2018-03-07:
I'm running into this now, wanting to pickle messages loaded from a bag, and hitting the flavour of this issue raised in #823.

Another possible angle for addressing the use case in question might be to simply implement the `__reduce__` method in the generated message classes so that as part of the pickling process, the message definition is stored, same as it is in bags— then when unpickling occurs, the message classes can be re-created as required.

kschwan, 2019-03-09:
> I'm running into this now, wanting to pickle messages loaded from a bag, and hitting the flavour of this issue raised in #823.
>
> Another possible angle for addressing the use case in question might [...]

This is perhaps a bit late, but I wrote a small script to re-write message objects using the system message definitions before saving to the pickle file: https://gist.github.com/kschwan/c32e057a9de01f2c04e8ac52adee7022 -- use at your own risk :)

closed 2025-05-31 by sloretz