ros/ros_comm #790: add_rostest ARGS only takes the first argument

efernandez, 2016-04-12:
`add_rostest(test/testlaunchfile.test argument1:=vaule1 argument2:=value2)`

when used in https://github.com/ros/ros_comm/blob/kinetic-devel/tools/rostest/cmake/rostest-extras.cmake.em#L55

generates

``` bash
-- run_tests.py: execute commands
  /opt/ros/indigo/share/rostest/cmake/../../../bin/rostest --pkgdir=/home/efernandez/ws/src/package_name --package=package_name --results-filename test_package_name__argument1_value1.xml --results-base-dir /home/efernandez/ws/build/package_name/test_results /home/efernandez/ws/src/package_name/test/testlaunchfile.test argument1:=value1
  argument2:=value2
```

Note that after the first argument, there's a new line. Maybe that's the problem.

efernandez, 2016-04-12:
It seems that `cmake_parse_arguments` is used properly, to take multiple arguments. So maybe the problem is when they're used to run the rostest command.

https://github.com/ros/ros_comm/blob/kinetic-devel/tools/rostest/cmake/rostest-extras.cmake.em#L18

https://cmake.org/cmake/help/v3.0/module/CMakeParseArguments.html

According to that link, _rostest_ARGS should have "argument1:=vaule1;argument2:=value2" for my example. Probably there's an issue with the `;`, which should be replaced with ` ` ?!

@dirk-thomas Any idea?

FYI @mikepurvis

dirk-thomas, 2016-04-12:
Please see #791 for the fix.

PR #791 (2016-04-12): fix passing multiple args to add_rostest
Fixes #790.
commit 73f81718 (2016-04-12): fix passing multiple args to add_rostest (fix #790)
commit 7f251de8 (2016-04-12): fix passing multiple args to add_rostest (fix #790)
closed 2016-04-12 by dirk-thomas