ros/ros_comm #794: roslaunch: new element proposal: <deprecated />

mikepurvis, 2016-04-14:
The purpose of this would be to notify users that the file they are launching is deprecated, probably with a message:

```
<deprecated message="This file has been superceded by xx/yy.launch, and will be removed in Kinetic." />
```

There could be an option to actually fail the launch.

```
<deprecated message="This file has been removed." level="fatal" />
```

It would also be possible to do this by just having a tiny node with a LOG_ERROR, etc, but having an actual tag for it would permit it to be flagged by roslaunch_add_file_check, for example.

peci1, 2016-05-24:
What is the difference to `<launch deprecated="message">` which has already been implemented?

mikepurvis, 2016-05-25:
There is none, and since making this ticket I discovered that attribute and have been using it. Thanks!

closed 2016-05-25 by mikepurvis