ros/ros_comm #798: [roslaunch] `env` tag does not affects `param command`

furushchev, 2016-04-26:
If given launch file like below:

```xml
<launch>
  <env name="TEST_ENV" value="1" />
  <node name="node1" pkg="foo" type="bar" />
  <param name="test_param" command="$(find foo)/scripts/bar" />
</launch>
```

where, node `bar` can read environment variable `TEST_ENV` by, for example in python `os.environ['TEST_ENV']`,
but script `bar` which is used in `param command` cannot read this variable.
How I pass environment variables to nodes used in `param command` without defining by `export` in shell beforehand?

dirk-thomas, 2016-04-26:
The documentation (http://wiki.ros.org/roslaunch/XML/env) says:

> When it is used inside of a <launch> tag, the <env> tag only applies to nodes declared after.

So I think the behavior you see is what is documented.

If you think making the environment variable available to the param command please consider creating a pull request for adding this feature. In that case I think the param tag should also support a nested env tag (the same as the node tag does).

Ermanas, 2019-11-01:
Hello, I fixed this problem with writing a simple .sh file. I set the environment variables on the .sh file and I also launched my .launch files on this .sh file.
**Important** : You have to run the .sh file with `source file.sh` or `. file.sh`. If you run your .sh file with `sh file.sh` it won't effect your current shell. ([credit](https://askubuntu.com/a/917344/968591))

My .sh file:

```
#!/bin/bash
echo
echo "Exporting ROS_MASTER_URI and ROS_HOSTNAME environment variables..."
export ROS_MASTER_URI="http://192.168.1.1:11311"
export ROS_HOSTNAME="192.168.1.1"
echo "-------------------------------"
echo "Evnvironment variables are set."
echo
echo "Launching the navigation stack"
echo "------------------------------"
roslaunch navigation_pkg ros_navigation.launch
```

closed 2025-05-31 by sloretz