ros/ros_comm #805: [rosbag] split bags do not all contain latched topics

wjwwood, 2016-05-17:
For example, if you start recording `/tf` and `/tf_static` (latched) and setup rosbag to split bags every so often, then only the first bag will contain the `/tf_static`. This is a bad UX because playing back just bags that aren't the first bag will not contain the latched topic.

One solution would be to put the latest message on each latched topic for each publisher in the new bag when splitting. The upside would be that each bag contains at least one message for each latched topic and for each publisher on it.

Another solution would be to add some option to `rosbag play` which allows rosbag to consider latched topics in other bags, but without playing them back. For example, if you have bags `1.bag`, `2.bag`, and `3.bag`, then you could run something like `rosbag play 2.bag --context-bags *.bag`. The expected behavior of this would be to only play messages from `2.bag`, but consider latched and advertised topics from any matched bag. If there was a single message published to a latched topic in `1.bag`, then the example command would still have that latched topic when playing back just `2.bag`.

dirk-thomas, 2016-05-17:
This looks like a duplicate of #706.

wjwwood, 2016-05-17:
Sure, I guess I'll close this and reference my description of the problem and possible solutions in the other issue.

closed 2016-05-17 by wjwwood