ros/ros_comm #808: rostopic/rqt_plot uses a large amount of memory if a large messages exists

TinyTheBrontosaurus, 2016-05-20:
See simple repeatable example on github:https://github.com/TinyTheBrontosaurus/rostopic_bug

In a nutshell, if a large message exists then simply advertising a small message causes 'rostopic hz' to use an inordinate amount of memory until it eventually fails.

SimpleMessage.msg
```
uint32 simple_variable_name
```

LargeMessage.msg (128MB array), but I can get it to happen at around 2.5MB
```
uint32[134217728] payload
```

Here's code that advertises the small message.
SuperSimple.cpp
```
#include <ros/ros.h>
#include <bug_report/SimpleMessage.h>

int main(int argc, char** argv) {
    ros::init(argc, argv, "super_simple_name");
    ros::NodeHandle nh;

    ros::Publisher mPublisher = nh.advertise<bug_report::SimpleMessage>("simple_topic", 1000);

    ros::Rate rate(1);

    for(int i = 0; i < 1000 && ros::ok(); i++) {
        rate.sleep();
        ROS_INFO("%d", i);
    }

    ROS_INFO("Done");
}
```

Removing the Big Message from CMakeLists.txt will fix the problem (but only if you manually delete `devel`. Calling `catkin_make clean` is not adequate on its own)

```
add_message_files(
        FILES
        SimpleMessage.msg
	# The Big Message breaks things. Comment out and remove devel
	# directory to fix
	BigMessage.msg
)

```

To reproduce, clone github.
Open 4 windows:
window 1) run top
window 2) run roscore
window 3) run clean_and_build.sh
window 3) then run run.sh
window 4) run hz.sh

See rostopic take a huge amount of RAM and then crash. Rostopic will continue to do this until roscore is reset. Also happens with rqt_plot (which is the first way I discovered this)

I'm using Ubuntu 15.04 desktop in a VirtualBox VM. Have 4GB RAM allocated to the VM. Running Jade.

spaghetti-, 2016-05-30:
verified on my end, I'll see if I can do anything about this without breaking backwards compatibility.

dirk-thomas, 2016-08-05:
Thank you for the report. After playing a bit with your great example I was able to reproduce it even with a single `rostopic pub /simple_topic bug_report/SimpleMessage 23` running.

Therefore I looked at the generate Python message `lib/python2.7/dist-packages/bug_report/msg/_BigMessage.py` (you might want to use `less` and not an editor... :wink:). Based on that I came up with the following fix to `genpy`: ros/genpy#64 That at least prevents the insane file size and with it the linear growth of the memory usage :partly_sunny:

But the `rostopic` process still uses roughly 3 GB memory (constantly) for my case...

dirk-thomas, 2016-08-05:
Thank you for the report. After playing a bit with your great example I was able to reproduce it even with a single `rostopic pub /simple_topic bug_report/SimpleMessage 23` running.

Therefore I looked at the generate Python message `lib/python2.7/dist-packages/bug_report/msg/_BigMessage.py` (you might want to use `less` and not an editor... :wink:). Based on that I came up with the following fix to `genpy`: ros/genpy#64 That at least prevents the insane file size and with it the linear growth of the memory usage :partly_sunny:

But the `rostopic` process still uses roughly 3 GB memory (constantly) for my case. This looks to be related to the `struct.Struct` being created in the generated message file...

dirk-thomas, 2016-08-05:
Thank you for the report. After playing a bit with your great example I was able to reproduce it even with a single `rostopic pub /simple_topic bug_report/SimpleMessage 23` running.

Therefore I looked at the generate Python message `lib/python2.7/dist-packages/bug_report/msg/_BigMessage.py` (you might want to use `less` and not an editor... :wink:). Based on that I came up with the following fix to `genpy`: ros/genpy#64 That at least prevents the insane file size and with it the linear growth of the memory usage :partly_sunny:

But the `rostopic` process still uses roughly 3 GB memory (constantly) for my case. Even when only importing the module in `ipython`. It looks to be related to the `struct.Struct` being created in the generated message file. ros/genpy#65 will solve the problem for this case by just deferring the initialization of `Struct` until it is actually being used.

dirk-thomas, 2016-08-08:
Addressed by ros/genpy#64 and ros/genpy#65.

closed 2016-08-08 by dirk-thomas