ros/ros_comm #810: console.h has compiler warnings

sytelus, 2016-05-24:
console.h doesn't compile if you turn on warnings as errors. I'm getting following warnings. This would prevent people from adding ROS headers who are turning on warnings as good practice.

Path: ros_comm/tools/rosconsole/include/ros/console.h

```
/opt/ros/indigo/include/ros/console.h:342:176: warning: format ‘%d’ expects argument of type ‘int’, but argument 8 has type ‘std::vector<dji_sdk::MissionWaypoint_<std::allocator<void> >, std::allocator<dji_sdk::MissionWaypoint_<std::allocator<void> > > >::size_type {aka long unsigned int}’ [-Wformat=]
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)

```

wjwwood, 2016-05-24:
This warning may be marked in our code, but if you read the message it is because a format string that contains a `%d` but was passed a `std::vector<dji_sdk::MissionWaypoint_<std::allocator<void> >, std::allocator<dji_sdk::MissionWaypoint_<std::allocator<void> > > >::size_type {aka long unsigned int} ` (which is not a type which that we define) for that argument position. Looking at the type, it might be a generated message type.

It's no different than if you had done something like:

```c++
std::vector<dji_sdk::MissionWaypoint_> vec;

```

I would need to know the full "traceback" of the warning to know if it is code we generated or code you wrote which is the root cause of the issue.

---

I agree we should fix warnings. In the future, you might consider opening a pull request if fixing the warning is tractable (it usually is).

We'll eventually get to reports of warnings, but it would be fixed faster with a pull request.

> This would prevent people from adding ROS headers who are turning on warnings as good practice.

I disagree that having warnings in a header prevents someone else from using it. I'd recommend that if you cannot quickly fix a warning in a dependencies header, that you disable the warning, see: https://gcc.gnu.org/onlinedocs/gcc/Diagnostic-Pragmas.html

You can wrap the include of ROS as a workaround like so (works for gcc and clang):

```c++
#ifndef _WIN32
# pragma GCC diagnostic push
# pragma GCC diagnostic ignored "-Wformat"
#endif
#include <ros/console.h>
#ifndef _WIN32
# pragma GCC diagnostic pop
#endif
```

But I don't recommend that unless you are ***100%*** sure it's a warning in our code and not a warning generated from how you use our code (which I believe to be the case here).

---

Can you give more details about the source of the warning or give us a minimal example which produces the warning? If you can't I'll eventually close this since we can't fix it with this amount of information. If you fix the error yourself, please share what you did and close the issue. Thanks!

evenator, 2016-05-24:
When I've seen this warning (and similar warnings), the cause is usually a `ROS_CONSOLE()` macro with the wrong flag. In this case, the flag specified was `%d` (signed int) when it should be `%zu` (`size_t`).

sytelus, 2016-05-25:
Thanks @evenator. That's indeed the case. I'' follow up with DJI folks.

wjwwood, 2016-05-25:
@sytelus can this this issue be closed then?

closed 2016-05-25 by sytelus