ros/ros_comm #813: rospy: get_num_connections() does not drop

naveedhd, 2016-05-31:
branch: Indigo

The number of connections of Publihser does not drop after the Subscriber connection is dropped.

This can be reproduced from following:
```
import rospy
from std_msgs.msg import String

def talker():
    publisher = rospy.Publisher("chatter", String, latch=True, queue_size=1)
    rospy.init_node('talker')
    changed = True
    while not rospy.is_shutdown():
        rospy.loginfo("Number of connections: %s" % publisher.get_num_connections())
        if changed:
            publisher.publish("dude")
            changed = False
        rospy.rostime.wallsleep(1.0)

if __name__ == '__main__':
    try:
        talker()
    except rospy.ROSInterruptException:
        pass
```

If I do `rostopic echo /chatter` and close it, the connections does not drop to 0:
```
[INFO] [WallTime: 1464685628.239426] Number of connections: 0
[INFO] [WallTime: 1464685629.240176] Number of connections: 0
[INFO] [WallTime: 1464685630.241219] Number of connections: 1
[INFO] [WallTime: 1464685631.243029] Number of connections: 1
[INFO] [WallTime: 1464685632.244549] Number of connections: 1
[INFO] [WallTime: 1464685633.246138] Number of connections: 1
```
The same issue occurs when using `queue_size=None`.

The similar issue has also been reported [here](https://github.com/ros/ros_comm/issues/526) . 

It seems that the connections are cleared only when `publish()` is called.

wkentaro, 2016-05-31:
Is this problem only with `latch=True` option for `rospy.Publisher`?
In that case, the issue title should be changed to make clear that.

spaghetti-, 2016-05-31:
I tried with `latch=False`, same issue.

naveedhd, 2016-06-01:
Same problem occurs with `latch=False`.

stonier, 2016-06-03:
Looks like right now the publisher garbage collects when trying to publish in [rospy/topics.py#L1076](https://github.com/ros/ros_comm/blob/kinetic-devel/clients/rospy/src/rospy/topics.py#L1076). 

I haven't delved deeper yet to find out why adding or removing connections isn't properly garbage collecting in [topics.py#L397](https://github.com/ros/ros_comm/blob/kinetic-devel/clients/rospy/src/rospy/topics.py#L397). Interesting comment from Ken there, though that doesn't actually match the continually increasing increment we see when repetitively doing rostopic echo/ctrl-c.

mgrrx, 2016-08-10:
This issue relates to #107. You are right, the garbage collection is only executed when publishing a message. I provided a patch for #107 in https://github.com/ros/ros_comm/pull/831 that also fixes the problem described here, at least for me.

flixr, 2018-08-25:
@naveedhd this is fixed with #831 so this issue can be closed.

closed 2018-08-25 by dirk-thomas