ros/ros_comm #820: rosbag: missing --node option from docs

Petrox, 2016-06-20:
--node=NODE           record all topics subscribed to by a specific node

is missing from

http://wiki.ros.org/rosbag/Commandline#record

dirk-thomas, 2016-06-27:
The wiki can be edited by anyone. Please consider to add the missing command line option to the page. If you are unsure you can post a draft here to get some feedback before.

Petrox, 2016-07-20:
Ok, I'll patch the wiki when I get permission to edit.

Petrox, 2016-07-28:
I've got the permission to edit, so it's fixed: http://wiki.ros.org/action/diff/rosbag/Commandline?action=diff&rev1=29&rev2=30

Congrats Dirk, github shows you didn't miss a day in the last 365 without a contribution. :) 

http://www.quickmeme.com/img/6b/6b8b3236f09f65ecccbdf9b42d8ccb662c2c6744c79b4219b16a3bef38a1da4c.jpg

closed 2016-07-28 by Petrox

dirk-thomas, 2016-07-28:
Thank you for updating the wiki.