ros/ros_comm #821: rosbag: create option to filter for topics AND nodes by hostname or IP

Petrox, 2016-06-20:
Take rostopic list, and iterate each, check rostopic info for subscribers and publishers.

If there is a publisher from the given host or IP, rosbag should record that topic.

The best solution would also filter the nodes, so it would not connect to the other nodes, just the ones matching the IP.

Use case:
 - having 20 computers on 10 robots
 - each of them could run a rosbag record --nodes-on=196.168.0.X (different IP)

So there would be 20 bag files, each containing every message that originates from the given host.

This would work with zero network overhead as rosbag would only connect to nodes on the local machine.

(rsyncing and even rosbag compressing could happen at a different time at a different computer)

dirk-thomas, 2016-06-27:
Please consider contributing a pull request which implements this feature.

Petrox, 2016-07-28:
I think this would require some "under the hood" tweaks into the subscription mechanics (not connecting to all nodes as the protocol would dictate), thats a little out of my scope currently, but this idea can stay here as a possible enhancement at later time. 
Also it might hurt some assumptions about the network topology, like when a new publisher arrives, and wants to connect to the listener (rosbag record), it should be refused friendly (the remote node should not fail)

So this needs some underground work. :)

closed 2025-05-31 by sloretz