ros/ros_comm #822: XML Args have odd behavior in groups

DLu, 2016-06-22:
I'm trying to do some conditional logic for setting args, and ran into some non-standard behavior for settings args in groups. 

## Working Base case
```
<launch>
    <arg name="plam" default="1" />
    <param name="foo" value="$(arg plam)" />        
</launch>
```
Result: 
```
PARAMETERS
 * /foo: 1
```
Okay so far. 

## Problematic Version 
```
<launch>
    <group>
        <arg name="plam" default="1" />
    </group>
    <param name="foo" value="$(arg plam)" />        
</launch>
```

Result: `[a_mere_test.launch] requires the 'plam' arg to be set`

Expected behavior: Same as the working example. 

I thought maybe the scope was different for inside the group tag, so I tried the next example....

## One more test
```
<launch>
    <arg name="plam" default="1" />
    <group>
        <arg name="plam" default="2" />
    </group>
    <param name="foo" value="$(arg plam)" />        
</launch>
```

Result: 
```Invalid <arg> tag: arg 'plam' has already been declared. 

Arg xml is <arg default="2" name="plam"/>
```

Is this the expected behavior?

dirk-thomas, 2016-06-27:
I am leaning towards saying, yes, this is "expected" behavior. The group tag will basically have it's own context. If an argument is only set in there it isn't visible in the outside scope (hence "Problematic Version"). But when the variable was set outside it is being passed into the context of the group which then complains about it being redeclared( hence "One more test").

DLu, 2016-07-01:
Fair enough.

closed 2016-07-01 by DLu

leolchat, 2017-01-25:
This is really counter intuitive, especially when mixing it with `if` and `unless` groups. (the behavior of adding the `if` tag to every command in the group is different than using a group.