ros/ros_comm #823: Messages loaded using rosbag python have incorrect type

felixduvallet, 2016-06-23:
I have found that `type(msg)` for a message loaded from a rosbag.Bag() does not match the actual type of the message.

For instance, a Float32 message created in a python module has type `<class 'std_msgs.msg._Float32.Float32'>`, but the same message loaded from a bag file has type `<class 'tmpfCH1Bk._std_msgs__Float32'>` (notice tmpfCH1Bk).

```
> Message from bag: data: 1.0. Type: <class 'tmpfCH1Bk._std_msgs__Float32'>
> New message: data: 1.0. Type: <class 'std_msgs.msg._Float32.Float32'>
> ...
> Message from bag: data: 2.0. Type: <class 'tmpfCH1Bk._std_msgs__Float32'>
> New message: data: 2.0. Type: <class 'std_msgs.msg._Float32.Float32'>
> [data: 1.0, data: 2.0]
```

A problem with this randomly-generated type occurs when pickling the messagess (i.e. rosbag -> pickle). After pickling, one cannot unpickle the messages since the module doesn't exist:

```
Traceback (most recent call last):
...
  File "/usr/lib/python2.7/pickle.py", line 1124, in find_class
    __import__(module)
ImportError: No module named tmpfCH1Bk
```
Note there's on exception: if you pickle and unpickle in the same instance of the interpreter, the temporary module is still defined, but that sortof goes against the point of pickling data to use later :).

Here is a MWE that reproduces this error.
https://www.dropbox.com/sh/i38kf379gmedsie/AACCe7AmdOjGCKk_aLfJmijYa?dl=0
Once the messages in the bag file get pickled, loading them from another run causes the ImportError described above.

dirk-thomas, 2016-06-23:
This sounds like a duplicate of #769.

closed 2016-06-27 by dirk-thomas

felixduvallet, 2016-06-28:
Yes, I agree. I'll have a look and comment/contribute over there.