ros/ros_comm #832: xmlrpcpp should use poll() for solving 1024 (FD_SETSIZE) clients problem in Linux

mgrrx, 2016-07-14:
xmlrpcpp uses select() for checking the open file descriptors, but this function can't check morethan FD_SETSIZE  file descriptors in Linux. FD_SETSIZE is set to 1024 by default and increasing that value would require to rebuild the current kernel.

select's 'successor' poll() isn't limited by 1024, so replacing select() by poll() would fix that problem.

kmactavish, 2017-07-28:
At the very least, xmlrpcpp should range-check the file descriptors and throw an exception if > 1024, rather than corrupt / buffer overflow the application.

For future googling, this can show up as
```c++
*** buffer overflow detected ***: /path/to/my/application
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x7329f)[0x7f4680f7829f]
/lib/x86_64-linux-gnu/libc.so.6(__fortify_fail+0x5c)[0x7f468100fbbc]
/lib/x86_64-linux-gnu/libc.so.6(+0x109a90)[0x7f468100ea90]
/lib/x86_64-linux-gnu/libc.so.6(+0x10ab07)[0x7f468100fb07]
/opt/ros/install/lib/libxmlrpcpp.so(_ZN6XmlRpc14XmlRpcDispatch4workEd+0xd8)[0x7f467c8ecf48]
/opt/ros/install/lib/libroscpp.so(_ZN3ros13XMLRPCManager16serverThreadFuncEv+0x139)[0x7f4682d329e9]
```

flixr, 2018-08-25:
@mgrrx this is fixed and can be closed, right?

dirk-thomas, 2018-08-27:
Assuming that #1133 addressed the problem by switching from `select` to `poll` I will close this.

closed 2018-08-27 by dirk-thomas