ros/ros_comm #836: Typ

mikepurvis, 2016-07-14:
We're seeing a bunch of crashes in our robots which take this form:

```
../../../../../ros-comm-d54e9be3421af71b70fc6b60a3bf916e779b43dc/clients/roscpp/src/libros/node_handle.cpp in ros::NodeHandle::subscribe at line 331

../../../../../ros-comm-d54e9be3421af71b70fc6b60a3bf916e779b43dc/clients/roscpp/src/libros/topic_manager.cpp in ros::TopicManager::subscribe at line 277

../../../../../ros-comm-d54e9be3421af71b70fc6b60a3bf916e779b43dc/clients/roscpp/src/libros/topic_manager.cpp in ros::TopicManager::registerSubscriber at line 485

../../../../../ros-comm-d54e9be3421af71b70fc6b60a3bf916e779b43dc/utilities/xmlrpcpp/src/XmlRpcValue.cpp in XmlRpc::XmlRpcValue::size at line 205

/usr/lib/x86_64-linux-gnu/libstdc++.so.6 in __cxa_throw ()

/usr/lib/x86_64-linux-gnu/libstdc++.so.6 in std::terminate() ()

/usr/lib/x86_64-linux-gnu/libstdc++.so.6 in ?? ()

/usr/lib/x86_64-linux-gnu/libstdc++.so.6 in __gnu_cxx::__verbose_terminate_handler() ()

abort.c in __GI_abort at line 89

../nptl/sysdeps/unix/sysv/linux/raise.c in __GI_raise at line 56
```

The line in question is here: https://github.com/ros/ros_comm/blob/d54e9be3421af71b70fc6b60a3bf916e779b43dc/utilities/xmlrpcpp/src/XmlRpcValue.cpp#L205

The cause is this logic which calls the master RPC without checking for the possibility of a `TypeInvalid` for the payload object: https://github.com/ros/ros_comm/blob/d54e9be3421af71b70fc6b60a3bf916e779b43dc/clients/roscpp/src/libros/topic_manager.cpp#L479-491

The simple fix is to just check for invalid and return false, similar to when `master::execute` itself returns false. However, my concern about this is that I don't really understand the underlying conditions which cause the RPC to fail. Should it be retried? Is there logic somewhere which already does this? AFAIK, the roscpp API makes no expectation of users to check for and retry subscriptions at the application layer.

FYI @jasonimercer

mikepurvis, 2016-07-15:
Following the stack from [the templated subscribe method](https://github.com/ros/ros_comm/blob/d54e9be3421af71b70fc6b60a3bf916e779b43dc/clients/roscpp/include/ros/node_handle.h#L400-L408):

```
  template<class M, class T>
  Subscriber subscribe(const std::string& topic, uint32_t queue_size, void(T::*fp)(M), T* obj, 
                       const TransportHints& transport_hints = TransportHints())
  {
    SubscribeOptions ops;
    ops.template initByFullCallbackType<M>(topic, queue_size, boost::bind(fp, obj, _1));
    ops.transport_hints = transport_hints;
    return subscribe(ops);
  }
```

To the [non-templated subscribe method that calls the TopicManager singleton](https://github.com/ros/ros_comm/blob/d54e9be3421af71b70fc6b60a3bf916e779b43dc/clients/roscpp/src/libros/node_handle.cpp#L331-L343):

```
  if (TopicManager::instance()->subscribe(ops))
  {
    Subscriber sub(ops.topic, *this, ops.helper);

    {
      boost::mutex::scoped_lock lock(collection_->mutex_);
      collection_->subs_.push_back(sub.impl_);
    }

    return sub;
  }

  return Subscriber();
```

If I'm reading this correctly, the TopicManager call returning `false` means that an uninitialized `Subscriber` instance will be returned by the `NodeHandle::subscribe` call, which will basically fail completely silently— you'll only really know if you interrogate that object.

mikepurvis, 2016-07-15:
Looks like there is some retry logic here: https://github.com/ros/ros_comm/blob/d54e9be3421af71b70fc6b60a3bf916e779b43dc/clients/roscpp/src/libros/master.cpp#L189-L239

Maybe the fix is as simple as changing line 202 of that file to be:

```
    if (!(b && b.valid()) && ok)
```

This would engage the existing retry logic in the face of a `TypeInvalid` response, however perhaps there are cases where `TypeInvalid` has legitimate semantic meaning, eg, improper API usage. Does xmlrpcpp give us some way of distinguishing a "try again" error vs. a "permanent failure" error?

mikepurvis, 2016-08-17:
Okay, I now have two instances of this crash with full core files.

Local variables from the `ros::TopicManager::registerSubscriber` frame:

```
i = 0
self_subscribed = <optimized out>
pub = {px = 0x1e72650, pn = {pi_ = 0x1e72630}}
sub_md5sum = <optimized out>
__PRETTY_FUNCTION__ = "bool ros::TopicManager::registerSubscriber(const SubscriptionPtr&, const string&)"
args = {static _doubleFormat = {static npos = <optimized out>, 
    _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, 
      _M_p = 0x1e5d6a8 "%.16g"}}, _type = XmlRpc::XmlRpcValue::TypeArray, _value = {asBool = 96, asInt = 31914336, 
    asDouble = 1.5767777027434545e-316, asTime = 0x1e6f960, asString = 0x1e6f960, asBinary = 0x1e6f960, asArray = 0x1e6f960, 
    asStruct = 0x1e6f960}}
result = {static _doubleFormat = {static npos = <optimized out>, 
    _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, 
      _M_p = 0x1e5d6a8 "%.16g"}}, _type = XmlRpc::XmlRpcValue::TypeInvalid, _value = {asBool = false, asInt = 0, asDouble = 0, 
    asTime = 0x0, asString = 0x0, asBinary = 0x0, asArray = 0x0, asStruct = 0x0}}
payload = {static _doubleFormat = {static npos = <optimized out>, 
    _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, 
      _M_p = 0x1e5d6a8 "%.16g"}}, _type = XmlRpc::XmlRpcValue::TypeInvalid, _value = {asBool = false, asInt = 0, asDouble = 0, 
    asTime = 0x0, asString = 0x0, asBinary = 0x0, asArray = 0x0, asStruct = 0x0}}
pub_uris = {<std::_Vector_base<std::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::basic_string<char, std::char_traits<char>, std::allocator<char> > > >> = {
    _M_impl = {<std::allocator<std::basic_string<char, std::char_traits<char>, std::allocator<char> > >> = {<__gnu_cxx::new_allocator<std::basic_string<char, std::char_traits<char>, std::allocator<char> > >> = {<No data fields>}, <No data fields>}, _M_start = 0x0, 
      _M_finish = 0x0, _M_end_of_storage = 0x0}}, <No data fields>}
```

And digging into `args` a little bit to understand the exact request which was made:

```
(gdb) print *(*args._value.asArray)._M_impl._M_start->_value.asString
$28 = {static npos = <optimized out>, 
  _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, 
    _M_p = 0x1e5fb88 "/t1_ekf_localization"}}

(gdb) print *((*args._value.asArray)._M_impl._M_start + 1)->_value.asString
$33 = {static npos = <optimized out>, 
  _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, 
    _M_p = 0x1e70198 "/platform_set_pose"}}

(gdb) print *((*args._value.asArray)._M_impl._M_start + 2)->_value.asString
$34 = {static npos = <optimized out>, 
  _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, 
    _M_p = 0x1e704d8 "geometry_msgs/PoseWithCovarianceStamped"}}

(gdb) print *((*args._value.asArray)._M_impl._M_start + 3)->_value.asString
$35 = {static npos = <optimized out>, 
  _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, 
    _M_p = 0x1e675a8 "http://CPR00612.local:33072/"}}
```

Observations:

- Both `result` and `payload` are TypeInvalid, while `args` appears to be a reasonable, sane request.
- `master::execute` clearly [returned true here](https://github.com/ros/ros_comm/blob/84eeb9ca1f5814d021d5f3bb6d3f8766e61a701c/clients/roscpp/src/libros/topic_manager.cpp#L479), since the exception was thrown on `payload.size()` a few lines down.
- I believe that XmlRpcClient::execute _should_ have [returned false here](https://github.com/ros/ros_comm/blob/kinetic-devel/utilities/xmlrpcpp/src/XmlRpcClient.cpp#L106-L107), in the face of an invalid response (via `parseResponse` returning false).

So I'm really struggling to see the code path where execute returns true, but the result is a TypeInvalid.

jasonimercer, 2016-08-17:
I had a look into this and found a code path that will result in an unhandled exception.

Consider the code at:

https://github.com/ros/ros_comm/blob/kinetic-devel/clients/roscpp/src/libros/master.cpp#L197

`b = c->execute(method.c_str(), request, response);`

if `response`is an array with 2 elements then it will get handed to 

https://github.com/ros/ros_comm/blob/kinetic-devel/clients/roscpp/src/libros/master.cpp#L228

`if (!XMLRPCManager::instance()->validateXmlrpcResponse(method, response, payload))`

This method ensures that response is either an array with 2 or 3 elements.

https://github.com/ros/ros_comm/blob/kinetic-devel/clients/roscpp/src/libros/xmlrpc_manager.cpp#L207

In the case that it is 2 elements a dummy payload is created (with an InvalidType) and true is returned

https://github.com/ros/ros_comm/blob/kinetic-devel/clients/roscpp/src/libros/xmlrpc_manager.cpp#L239

this payload then find its way to `payload.size()` at:

https://github.com/ros/ros_comm/blob/kinetic-devel/clients/roscpp/src/libros/topic_manager.cpp#L485

and will throw an exception.

I'm not sure how you end up with an array with 2 elements (or any data for that matter) but the fact that a check is made for 2 elements seems to indicate it's not an unreasonable value.

dirk-thomas, 2016-08-18:
What makes you think that the following value is of type *invalid*?

```
XmlRpcValue("<value><array><data></data></array></value>", 0)
```

I would expect the trace to look like this:

* constructor: https://github.com/ros/ros_comm/blob/7e848445a8d8f8b4acef5fb9cdda4d490536a288/utilities/xmlrpcpp/include/XmlRpcValue.h#L69-L70
* parse the passed xml: https://github.com/ros/ros_comm/blob/7e848445a8d8f8b4acef5fb9cdda4d490536a288/utilities/xmlrpcpp/src/XmlRpcValue.cpp#L216
* find the value tag: https://github.com/ros/ros_comm/blob/7e848445a8d8f8b4acef5fb9cdda4d490536a288/utilities/xmlrpcpp/src/XmlRpcValue.cpp#L221
* find the array tag: https://github.com/ros/ros_comm/blob/7e848445a8d8f8b4acef5fb9cdda4d490536a288/utilities/xmlrpcpp/src/XmlRpcValue.cpp#L239
* find the data tag: https://github.com/ros/ros_comm/blob/7e848445a8d8f8b4acef5fb9cdda4d490536a288/utilities/xmlrpcpp/src/XmlRpcValue.cpp#L487
* set the value to by of type array with zero items: https://github.com/ros/ros_comm/blob/7e848445a8d8f8b4acef5fb9cdda4d490536a288/utilities/xmlrpcpp/src/XmlRpcValue.cpp#L490

Which should result in an array value type.

mikepurvis, 2016-08-18:
@dirk-thomas I can confirm that with the following test program:

```
#include "XmlRpcValue.h"
#include <iostream>

void main()
{
    std::string empty_array = "<value><array><data></data></array></value>";
    int offset = 0;
    XmlRpc::XmlRpcValue payload(empty_array, &offset);
    std::cout << payload.valid() << payload.toXml();  // "1<value><array><data></data></array></value>"
}
```

However this also shows that the default state for an XmlRpcValue is TypeInvalid:

```
#include "XmlRpcValue.h"
#include <iostream>

void main()
{
    XmlRpc::XmlRpcValue payload;
    std::cout << payload.valid();  // "0"
}
```

Maybe there's some pathway through these functions where `payload` never gets set at all?

jasonimercer, 2016-08-22:
Yes, I was mistaken (and then on vacation). I misread that std::string, int* constructor. That constant string should result in an array rather than an Invalid type.

mikepurvis, 2017-02-14:
We haven't seen new crashes like this in several weeks. Going to call this one closed, almost certainly by #938.

closed 2017-02-14 by mikepurvis