ros/ros_comm #845: rosgraph - roscore crash on OSX, depending on network setup

luke-gumbley, 2016-07-23:
Crashes for me with a fresh Yosemite install. For a similar report, see http://answers.ros.org/question/220631/roscore-bug-errno8/

The issue is caused by OSX being categorised as non-unix-like in network.py, meaning the less sophisticated local address lookup is used. PR inbound.

dirk-thomas, 2016-07-25:
The references questions seems to have been resolved by resetting the environment variables and setting the hostname. Can you please describe what your exact problem is?

luke-gumbley, 2016-07-25:
On OSX, the machine name is not automatically resolvable to the local machine. You must either append ".local" or add the machine name to the hosts file as 127.0.0.1.

In the referenced example, the initial report did have problems with environment variables but the follow-up comment beneath refers to the actual issue ("my computer couldn't ping itself under its own name").

Resolution is straightforward - rosgraph already includes a more sophisticated method of obtaining local addresses, but it's not used as the OS check for it excludes OSX. My PR just adds OSX back in.

dirk-thomas, 2016-08-05:
Addressed by #846.

closed 2016-08-05 by dirk-thomas