ros/ros_comm #868: 'rostopic echo' returning incomplete message data for some topics

davidepalozzi, 2016-08-16:
it appears that the kinetic-devel branch has a regression, introduced at commit https://github.com/ros/ros_comm/commit/e0f63978fe0dd239ba8002a7e79756cdf1a335d0

`rostopic echo` may not properly return the message data for some topics

this can be reproduced as follows:

* put the latest ros_comm (kinetic-devel branch) in a source overlay workspace (i overlaid this on-top of indigo)
* start a roscore 
* start the sample TF broadcaster node (code below)
* run `rostopic echo /tf`. this will return `transforms: None`
* spinning up a rosnode (tested via rospy) and subscribing to the topic does indeed result in the expected message. this is only a problem using `rostopic echo`

repeating the above on any previous commits (i.e. 9ff6955679afc791240aba637dacde6eeb5ae8eb) of ros_comm has no issue and `rostopic echo /tf` returns the expect result

sample TF broadcaster node:
    #!/usr/bin/env python  
    import rospy
    import tf
    import time
    
    if __name__ == '__main__':
        rospy.init_node('test')
    
        while not rospy.is_shutdown():
            br = tf.TransformBroadcaster()
            br.sendTransform((0, 0, 0),
                             tf.transformations.quaternion_from_euler(0, 0, 0),
                             rospy.Time.now(),
                             'test',
                             "world")
            time.sleep(1)

dirk-thomas, 2016-08-16:
Thank you for reporting this. The change was part of #724.

@wkentaro Can you please try to reproduce the reported problem and if necessary provide a patch.

wkentaro, 2016-08-23:
I think https://github.com/ros/ros_comm/pull/872 fixes this problem.

dirk-thomas, 2016-08-23:
@wkentaro Thank you for the fix.

@davidepalozzi Can you please try the proposed fix from the PR and comment here if it resolves the your problem.

davidepalozzi, 2016-08-26:
yep, looks good! Thanks guys.

dirk-thomas, 2016-08-30:
@davidepalozzi Thanks for reporting and checking the patch.

@wkentaro Thanks for the fast fix.

PR #872 (2016-08-30): [rostopic] Fix rostopic echo for list of ROS message to fix #868
Fix #868

The output will be like below:

```
% rostopic echo /tf
transforms: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1471928060
        nsecs: 475493655
      frame_id: world
    child_frame_id: base_link
    transform: 
      translation: 
        x: 0.0
        y: 0.0
        z: 0.0
      rotation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0

% rostopic echo /tf --noarr
transforms: <array type: geometry_msgs/TransformStamped, length: 1>

% rostopic echo /tf --nostr
transforms: 
  - 
    hea [... cut]
commit 3093ddc4 (2016-08-30): Merge pull request #872 from wkentaro/fix-rostopic-echo-868

[rostopic] Fix rostopic echo for list of ROS message to fix #868
commit 662d5dfd (2016-08-30): Fix rostopic echo for list of ROS message to fix #868

Fix #868
closed 2016-08-30 by dirk-thomas