ros/ros_comm #876: Different output of rosconsole message using ROSCONSOLE_FORMAT environmental variable

wkentaro, 2016-08-25:
Refering to http://wiki.ros.org/rosconsole#Console_Output_Formatting, it says the default value is
```
export ROSCONSOLE_FORMAT='[${severity}] [${time}]: ${message}'
```

But the outputs are different
```
% export ROSCONSOLE_FORMAT='[${severity}] [${time}]: ${message}'

% python spam.py 
[INFO] [1472113962.464240]: a
[WARN] [1472113962.464692]: a
[FATAL] [1472113962.465004]: a
[ERROR] [1472113962.465293]: a
[INFO] [1472113962.465572]: a
[WARN] [1472113962.465838]: a
[ERROR] [1472113962.466106]: a
[FATAL] [1472113962.466405]: a
[INFO] [1472113962.466765]: a
[WARN] [1472113962.467037]: a
[ERROR] [1472113962.467304]: a
[FATAL] [1472113962.467577]: a
```

```
% unset ROSCONSOLE_FORMAT 

% python spam.py         
[INFO] [WallTime: 1472113988.890666] a
[WARN] [WallTime: 1472113988.891116] a
[FATAL] [WallTime: 1472113988.891413] a
[ERROR] [WallTime: 1472113988.891683] a
[INFO] [WallTime: 1472113988.891950] a
[WARN] [WallTime: 1472113988.892203] a
[ERROR] [WallTime: 1472113988.892459] a
[FATAL] [WallTime: 1472113988.892712] a
[INFO] [WallTime: 1472113988.893018] a
[WARN] [WallTime: 1472113988.893272] a
[ERROR] [WallTime: 1472113988.893521] a
[FATAL] [WallTime: 1472113988.893764] a
```

dirk-thomas, 2016-08-30:
The default value in C++ is as mentioned in the wiki: https://github.com/ros/ros_comm/blob/cc808bd9a266149f2fffd46b623435927e719d20/tools/rosconsole/src/rosconsole/rosconsole.cpp#L108

> [${severity}] [${time}]: ${message}

In Python support for the formatting was implemented in #519. If the environment variable is not set the patch maintained the previous behavior which included the `WallTime: ` string.

I am not sure if it is worth changing the default behavior in Python or if the wiki page should just mention it.

dirk-thomas, 2016-08-30:
I updated the wiki page to mention the different behavior: http://wiki.ros.org/action/diff/rosconsole?action=diff&rev1=70&rev2=71

I will close this ticket for now. If you think it is worth unifying both behaviors to be identical please consider to create a pull request.

closed 2016-08-30 by dirk-thomas