ros/ros_comm #881: roslaunch leave processes around

asmodehn, 2016-08-29:
Current behavior on signal for linux is : 
- forward SIGINT
- wait for process shutdown
- send SIGTERM
- exit.

However it is possible for a process to stay alive after SIGTERM.

To avoid issues, ( like old process conflicting with new process when restarting ) we should ensure that roslaunch cleanup all the processes it is managing before shutting down.

=> How about triggering SIGKILL if SIGTERM fails to clean up all roslaunch'ed children ?
supervisord already does that : http://supervisord.org/subprocess.html

tfoote, 2016-08-29:
roslaunch does escalate from SIGINT to SIGTERM to SIGKILL Can you provide a reproducible example of leaving child processes? 

roslaunch only exits after a SIGTERM if it gets a return code. Otherwise it continues to escalate to a SIGKILL: https://github.com/ros/ros_comm/blob/98aa98e565ea623aae61c02f54ecb06c7ee190dd/tools/roslaunch/src/roslaunch/nodeprocess.py#L407-L422

asmodehn, 2016-08-31:
Ah, seems I was mistaken, thanks for the reference.

In my case I am using celery workers and web processes (tornado) which sometimes do not get cleaned up... It could also be because the multiple layer of process management...

I ll try to come up with a simpler example that reproduce the problem with roslaunch.

asmodehn, 2017-06-12:
Update : I haven't and probably wont have any time to try and reproduce this...
I suspect it was due to the multiple level of process management, shells calls, and python calling python... so probably not related with roslaunch itself.
Feel free to close.

k-okada, 2017-06-12:
It seems you had https://github.com/ros-infrastructure/catkin_pkg/issues/175#issuecomment-303924068 issues.

dirk-thomas, 2017-07-07:
I will close this for now. Please feel free to comment with a reproducible example and I am happy to reopen this.

closed 2017-07-07 by dirk-thomas