ros/ros_comm #883: rosbag::Recorder: Less than  of space free on disk with FILENAME.  Disabling recording.

dborrman, 2016-09-01:
min_space and min_space_str should be added to the default constructor of rosbag::Recorder, using the old hard-coded values  for 1GB, to stop old applications from terminating on newer ROS version.

When not setting those values explicitly the check for free space fails, the recorder terminates and the following error message with an empty string is not helpful:

ROS_ERROR("Less than %s of space free on disk with %s.  Disabling recording.", options_.min_space_str.c_str(), bag_.getFileName().c_str());

dirk-thomas, 2016-09-01:
#884 should address the problem you described. It would be great if you could try the fix and confirm that it works for you.

closed 2016-09-01 by dirk-thomas

dborrman, 2016-09-01:
Yes, this works for me. Thanks!

PR #884 (2016-09-01): set default values for min_space and min_space_str
Fixes #883.