ros/ros_comm #889: roslaunch: pass_all_args exception if there are extra args defined

scpeters, 2016-09-07:
I just tried using this attribute on a fairly complex launch file that has multiple `<arg>`s declared, and it leads to an exception because not all of the `<arg>`s are defined in the `<include>`d file.

For example, in the `drcsim_gazebo` package, [atlas.launch](https://bitbucket.org/osrf/drcsim/src/194be8500fef81593f79607a21ee2badd9700a0e/drcsim_gazebo/launch/atlas.launch?at=default&fileviewer=file-view-default#atlas.launch-14) includes [atlas_no_controllers.launch](https://bitbucket.org/osrf/drcsim/src/194be8500fef81593f79607a21ee2badd9700a0e/drcsim_gazebo/launch/atlas_no_controllers.launch?at=default&fileviewer=file-view-default) and passes three arguments. I removed some of the duplicate argument definitions in [this commit](https://bitbucket.org/osrf/drcsim/commits/d5c93d7db649ca4df6c07d6fcba6f9e77953913a), but roslaunch gives an exception since there are extra `<arg>`s in `atlas.launch` that aren't part of `atlas_no_controllers.launch` (for example, [inertia_args](https://bitbucket.org/osrf/drcsim/src/194be8500fef81593f79607a21ee2badd9700a0e/drcsim_gazebo/launch/atlas.launch?at=default&fileviewer=file-view-default#atlas.launch-8)).

I'm guessing this will be close as a "won't fix", but I wanted to mention it since I was excited when @gerkey added this feature in #710, but now there's quite a few instances when I won't be able to use it.

gerkey, 2016-09-07:
Oh, that's interesting. I hadn't considered that case. Can you tell me which exception is thrown, to help me track down the code path that leads to it and have a look at what can be done?

scpeters, 2016-09-07:
~~~
[roslaunch][INFO] 2016-09-06 17:25:55,495: Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2016-09-06 17:25:55,518: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2016-09-06 17:25:55,518: roslaunch starting with args ['/opt/ros/indigo/bin/roslaunch', 'drcsim_gazebo', 'atlas.launch', 'gzname:=gzserver']
[roslaunch][INFO] 2016-09-06 17:25:55,519: roslaunch env is ...
[roslaunch][INFO] 2016-09-06 17:25:55,519: starting in server mode
[roslaunch.parent][INFO] 2016-09-06 17:25:55,519: starting roslaunch parent run
[roslaunch][INFO] 2016-09-06 17:25:55,519: loading roscore config file /opt/ros/indigo/etc/ros/roscore.xml
[roslaunch][INFO] 2016-09-06 17:25:55,607: Added core node of type [rosout/rosout] in namespace [/]
[roslaunch.config][INFO] 2016-09-06 17:25:55,608: loading config file /data_fast/scpeters/ws/drcsim/install/share/drcsim_gazebo/launch/atlas.launch
[roslaunch][INFO] 2016-09-06 17:25:55,638: Added node of type [drcsim_gazebo/run_gzserver] in namespace [/]
[roslaunch][ERROR] 2016-09-06 17:25:55,638: unused args [hand_suffix, inertia_args, model_args] for include of [/data_fast/scpeters/ws/drcsim/install/share/drcsim_gazebo/launch/atlas_no_controllers.launch]
[roslaunch][ERROR] 2016-09-06 17:25:55,638: The traceback for the exception was written to the log file
[roslaunch][ERROR] 2016-09-06 17:25:55,639: Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/__init__.py", line 307, in main
    p.start()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/config.py", line 455, in load_config_default
    raise RLException(e)
RLException: unused args [hand_suffix, inertia_args, model_args] for include of [/data_fast/scpeters/ws/drcsim/install/share/drcsim_gazebo/launch/atlas_no_controllers.launch]

[rospy.core][INFO] 2016-09-06 17:25:55,639: signal_shutdown [atexit]
~~~

tfurf, 2017-03-02:
I'm also having this problem, and with nested launch files, it makes the `pass_all_args` unfortunately useless, where it should be able to greatly reduce the amount of repeated text all over the place.

gerkey, 2017-03-03:
Thanks for the bump. I'll look into this.

luator, 2018-03-28:
Any new on this? I actually never understood why it throws an exception just because some arguments are not used.

alextoind, 2018-03-29:
This workaround excludes the check when `pass_all_args` is set (https://github.com/ros/ros_comm/blob/2c8a3f65628d0362d97c29d333d5669b2c7803a6/tools/roslaunch/src/roslaunch/xmlloader.py#L630-L631):

```
            if not pass_all_args:
                # check for unused args
                loader.post_process_include_args(child_ns)
```

I've not created any pull request yet since I don't know if this behavior is what you want.

alextoind, 2018-06-13:
@scpeters @gerkey, if you guys give me some feedback about the intended behavior I can make a PR.

GoosebumpsFactory, 2018-10-13:
> @scpeters @gerkey, if you guys give me some feedback on the intended behavior I can make a PR.

This is precisely the behavior that would reduce the verbosity of many of my launch files. Is there a reason this behavior wasn't made PR?

alextoind, 2018-10-13:
@GoosebumpsFactory I think there's no much interest, but I can make the PR anyway and discuss implementation details there.

GoosebumpsFactory, 2018-10-14:
@alextoind Thanks. That would be great.

closed 2018-11-20 by dirk-thomas

mogumbo, 2020-04-08:
Has anyone looked at this lately? I would also use pass_all_args quite a bit if the unused args check was skipped. Because that check happens, I have never been able to use it.

gerkey, 2020-04-08:
Looks like @alextoind fixed it in #1520 .

mogumbo, 2020-04-08:
Thanks. I should have read the thread more carefully. So now I just have to figure out why I'm still getting the unused arg error. Maybe it's some separate issue.

alextoind, 2020-04-08:
@mogumbo it is not been backported to ROS Kinetic yet, so that might be the problem.
Actually, I think that it will not be brought to ROS Kinetic at this point.

mogumbo, 2020-04-08:
I'm using melodic. I have ros-melodic-roslaunch 1.14.3-0bionic.20191211.003455 and /opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py does not contain @alextoind's change. The change exists in the melodic-devel source in this repo, and my package is dated over a year after the change was made. I'm confused. Is my package not created from melodic-devel?

P.S. There is a newer package available, but my organization doesn't give me install privileges and hasn't been able to update anything on my computer since quarantine started.

mogumbo, 2020-04-08:
Okay, I got an explanation for this. The package date is recent, but 1.14.3 predates the merge. I need to get my organization to update me and my group to 1.14.5. Sorry for all the static on this thread.

commit b1713cf2 (2018-11-20): Fix #889 : Exclude unused args check if pass_all_args is set (#1520)

* Exclude unused args check if pass_all_args is set

* Add test for pass_all_args fix
commit 317902f8 (2020-08-03): Fix #889 : Exclude unused args check if pass_all_args is set (#1520)

* Exclude unused args check if pass_all_args is set

* Add test for pass_all_args fix