ros/ros_comm #891: ros::init with no __name and name = "" is not tested in roscpp?

furushchev, 2016-09-08:
I'd like to report a possible bug at roscpp initialization.
In roscpp initializer `ros::init (const M_string &remappings, const std::string &name, uint32_t options=0)`, a node can be initialized even empty string is passed to `name` argument if remappings are passed through from command line arguments.
But if `__name:=foobar` is removed from `remappings` and `name` is empty, ros node can be registered with invalid name, which can occur unexpected errors.
I did experiment in some conditions. the result is followings:

| arg remap __name | arg name |  group | this_node::name | this_node::namespace |
|------------------|------------|------|----------------|---------------|
| test_node | test_node |   |  /test_node | / |
| test_node | test_node |  group1 |  /group1/test_node | //group1 |
| test_node |  |   |  /test_node | / |
| test_node |  |  group1 |  /group1/test_node | //group1 |
|  | test_node |   |  /test_node | / |
|  | test_node |  group1 |  /group1/test_node | //group1 |
|  |  |   |  / | / |
|  |  |  group1 |  //group1 | //group1 |

In the last 2 conditions, namespace and name are the same and other nodes cannot resolve the node.

furushchev, 2016-09-08:
test code: 

```c++
#include <ros/ros.h>
#include <map>

int main(int argc, char** argv) {
  bool remap_name = true;
  bool node_name_empty = true;
  std::map<std::string, std::string> remaps;
  if (remap_name) remaps["__name"] = "test_node";
  std::string node_name = "test_node";
  if (name_empty) node_name = "";
  ros::init(remaps, node_name, (uint32_t)0);

  ROS_INFO_STREAM("getName: " << ros::this_node::getName());
  ROS_INFO_STREAM("getNameSpace: " << ros::this_node::getNamespace());

  return 0;
}
```

```xml
<launch>
  <group ns="group1">
    <node name="nstest_node" pkg="roscpp_nstest" type="nstest" output="screen" />
  </group>
</launch>
```

furushchev, 2016-09-08:
```bash
$ rostopic info /group1/string
Type: std_msgs/String

Publishers: 
 * //group1 (unknown address //group1)

Subscribers: None
```

furushchev, 2016-09-08:
In the first place, is it allowed to initialize node with empty string as `name` argument in `ros::init``?

dirk-thomas, 2016-09-08:
Since node names are a [graph resource name](http://wiki.ros.org/Names#Valid_Names) I would expect an empty string to be invalid.

furushchev, 2016-09-09:
@dirk-thomas Thank you for clarifying! OK. Then I'll create the pull request in which passing empty string to `node_name` is invalid.

furushchev, 2016-09-09:
(Could I wait other some mention about this issue for some days?)

dirk-thomas, 2016-09-09:
@ros/ros_team What do you think?

tfoote, 2016-09-09:
Empty strings are definitely invalid names and should error either if passed directly or remapped to an empty value.

wjwwood, 2016-09-10:
I second what @tfoote said.

furushchev, 2016-09-11:
@ros/ros_team Thank you all for advice! I created pull requests.

furushchev, 2016-09-22:
close via https://github.com/ros/ros_comm/commit/bd3af70520648783da8aa4d3610f234a4d2bd41f

closed 2016-09-22 by furushchev