ros/ros_comm #900: deprecate old spinner behavior

rhaschke, 2016-09-20:
In #867 a new scheme for monitoring access of spinners to callback queues was proposed, ensuring that only a single single-threaded spinner can access a given callback queue.
To maintain backwards compatibility to already released ROS distros, it was decided to keep the old, non-strict error handling. In a new ROS distro (L-turtle) a more strict error checking should be applied, i.e. throwing an exception if starting of the spinner failed.
To this end, commit 91be0e5 needs to be reverted. This will also re-enable some more unittests, which check for throwing of these exceptions.

mikepurvis, 2017-02-14:
@rhaschke Would you be willing to put up a PR for this so that @dirk-thomas can merge it as part of preparing the Lunar release of ros_comm?

rhaschke, 2017-02-14:
I will be happy to do so. However, I think we should have a lunar-devel branch for this, shouldn't we?

dirk-thomas, 2017-02-14:
You can just go ahead and create a patch / PR against kinetic-devel. Once the new branch exists we can easily retarget the PR.

mikepurvis, 2017-02-15:
Closed in favour of #988.

closed 2017-02-15 by mikepurvis