ros/ros_comm #901: genpy: does not validate fixed array length

trainman419, 2016-09-21:
The serialization code generated by genpy does not check that a fixed array is the declared size; it just blindly serializes the array as given.

In addition, the rospy deserializer gets stuck on this input but does not print a user-visible error message, so it's very difficult to tell that this has occurred.

This can be reproduced with this package: https://github.com/trainman419/rospy_fixed_len_bug

First run rostopic echo:

    rostopic echo /fixed_array

And then run the node that demonstrates the bug:

    rosrun rospy_fixed_len_bug pub.py

Observe that rostopic prints the first message (which is valid) and then hangs when it receives the invalid message. Also observe that the publisher does not print any error messages and appears to keep publishing messages, even though the subscriber is not receiving them.

Note that this only occurs for generated types (like `std_msgs/Int32`); the native types (like `int32`) have proper bounds checking and throw an exception when attempting to publish a array with the wrong length.

ernestmc, 2018-04-17:
I can confirm this bug also happens for native strings.

Test message:
```
# m1.msg
string[3] string_array
```

Test code:
```
import rospy
from test.msg import m1

rospy.init_node("test")
pub = rospy.Publisher("/test_topic", m1, queue_size=10)
rate = rospy.Rate(1)

while not rospy.is_shutdown():
    good_msg = m1()
    good_msg.string_array = ["string_1", "string_2", "string_3"]
    pub.publish(good_msg)

    bad_msg = m1()
    bad_msg.string_array = ["string_1", "string_2"]
    pub.publish(bad_msg)

    rate.sleep()
```

dirk-thomas, 2019-01-26:
Addressed by ros/genpy#92.

closed 2019-01-26 by dirk-thomas