ros/ros_comm #902: Support messages over 1 GB in size

calderpg, 2016-09-22:
Subscribers and services are limited to 1000000000-byte messages, even though messages up to 4294967295 bytes are possible. Such large messages are useful for sending large data structures like voxel grids and pointclouds. Right now, these messages can be serialized and sent, but will cause the recipient to close the connection when they are received.

This limit is enforced in [service servers](https://github.com/ros/ros_comm/blob/be7c7e03338eb34266b06a2a22b6118f6e3d7ee1/clients/roscpp/src/libros/service_server_link.cpp#L199), [service clients](https://github.com/ros/ros_comm/blob/be7c7e03338eb34266b06a2a22b6118f6e3d7ee1/clients/roscpp/src/libros/service_client_link.cpp#L191), [topics](https://github.com/ros/ros_comm/blob/be7c7e03338eb34266b06a2a22b6118f6e3d7ee1/clients/roscpp/src/libros/transport_publisher_link.cpp#L170), [connection headers](https://github.com/ros/ros_comm/blob/be7c7e03338eb34266b06a2a22b6118f6e3d7ee1/clients/roscpp/src/libros/connection.cpp#L394), and there may be others I haven't seen.

Ideally this limit would be increased, or at least controlled by a single parameter like an environment variable or compile-time define.

matt-attack, 2018-07-11:
I am running into this issue as well and accidentally opened a duplicate issue. https://github.com/ros/ros_comm/issues/1453

matt-attack, 2018-11-19:
@dirk-thomas Is there any reason that raising this limit isn't reasonable? I would think raising it to match the protocol limits would make sense.

dirk-thomas, 2018-11-19:
> Is there any reason that raising this limit isn't reasonable?

I don't see a reason why it shouldn't be possible to raise the limit.

gurselturkeri, 2023-01-27:
any update on this topic?