ros/ros_comm #903: [rostest] Randomize Gazebo port number

mikepurvis, 2016-09-22:
rostest has some nifty logic which allows rostests to run on top of each other, by randomizing the rosmaster's port number:

https://github.com/ros/ros_comm/blob/98aa98e565ea623aae61c02f54ecb06c7ee190dd/tools/rostest/src/rostest/rostest_parent.py#L84

Unfortunately, there can still be collisions with concurrent tests when Gazebo is involved, as there will be contention over the default GAZEBO_MASTER_URI value of `http://localhost:11345`. Would it be reasonable to speculatively set GAZEBO_MASTER_URI in the test environment, with a known-available port?

dirk-thomas, 2016-09-22:
A short detour on how the random port number for `rostest` works: the feature of `rostest` to choose a random port for the ROS master was add in #468. It is achieved by creating the `Master` instance with a port `0` which will then let the master pick a random port. After the master is started and knows its port number each executable will get the chosen ROS master uri set in the environment: https://github.com/ros/ros_comm/blob/4468f862d3c80dbba1bb61b5a8cc6c9a86de641b/tools/roslaunch/src/roslaunch/core.py#L201

I understand your use case with running multiple tests using Gazebo. On the other hand nothing in these ROS packages knows about Gazebo at the moment. And the same request could be made for almost any other external software which needs a specific environment in order to run multiple instances in parallel. Therefore I hesitate to add something specifically for Gazebo. A "generic" feature which enables your use case for Gazebo but also covers similar use cases would be nicer.

I am not sure how this can be achieved best. Since Gazebo is run from one entry in the roslaunch file it is a bit different than the ROS master described above.

* Do any of the other nodes in the launch file also need the Gazebo uri?
  * If no, would it be an option to let Gazebo pick its own port (will require a change to Gazebo)?
  * If yes, the env var needs to be set externally. And the chosen port number must be unique and available.

mikepurvis, 2016-09-23:
My primary interest is in running tests concurrently from multiple repos on the same buildslave (without going to Docker or whatever for port isolation), or by a parallel catkin_tools invocation (eg https://github.com/catkin/catkin_tools/issues/397). So from either of those perspectives, it would be perfectly possible to just have a `gazebo_testing` package with an env hook that picks a random available port and just sets the variable.

However, that doesn't help the `catkin_make run_tests` case, since it would still be a single `GAZEBO_MASTER_URI` value. For that there'd need to be some kind of extra pre-test environment injection. Unsure on the best way to achieve that.

I agree in principal that it sucks to put something Gazebo-specific like this into rostest, but the complexity of a pluggable solution seems really high; given that it's just setting a var without any _actual_ dependence, I wonder if it would be worth it.

mikepurvis, 2016-09-23:
That would be picking the port number at configure time rather than execution time, wouldn't it?

The idea of `add_rostest_with_gazebo()` is appealing, but I feel like it would need a little bit deeper integration than just wrapping `add_rostest()`.

dirk-thomas, 2016-09-23:
That is true.

The beauty with ROS master is that it picks the port on its own before any node from the launch file is started. And then that can be used to set the environment in all nodes. I don't see yet how something similar can be done with Gazebo (in a way that ensure collision free port numbers).

mikepurvis, 2016-09-24:
Yes, in the case of setting up the env var in advance, it will still just be a random number best effort. Without cooperation from Gazebo, that's the best that's possible.

dirk-thomas, 2016-09-26:
Since Gazebo is only one process in a launch file I don't see a good way how even with the support of Gazebo this could be improved.

@ros/ros_team Any ideas how to approach this?

mikepurvis, 2016-09-26:
It would have to be a scheme where `add_gazebo_rostest()` launches a standalone gazebo instance (similar to the roscore), and passes its port number into the environment of the launch file with the actual tests, which expect to interact with a supplied instance rather than launching their own.

In the short term, I still think there's value in having an option to at least randomize it, or do some kind of round-robin scheme.

wjwwood, 2016-09-26:
Setting the env variable to control the port Gazebo uses and randomizing that is the best option of which I can think.