ros/ros_comm #905: roscpp -> rospy topic subscription fails when unreliable transport hint is provided

guillaumeautran, 2016-09-26:
In roscpp, subscribing to a topic specifying the unreliable protocol hint does not fallback to a reliable transport if the publisher does not support unreliable transport. 

If a roscpp-based node attempts to subscribe to a topic published by a rospy-based node and specify the 'unreliable' transport hint, the topic connection does not get established and the protocol negotiation does not fall back to a reliable protocol as suggested by the documentation.

Documentation here: [Overview Publishers and Subscribers](http://wiki.ros.org/roscpp/Overview/Publishers%20and%20Subscribers#Transport_Hints) clearly state that:
   `Also, these transport options are just hints -- if, for example, the publisher on the topic you're subscribing to does not support an unreliable connection, the connection will still be made, but it will use a reliable connection instead (e.g. TCP). `

Code: [subscription.cpp#L351](https://github.com/ros/ros_comm/blob/indigo-devel/clients/roscpp/src/libros/subscription.cpp#L351)

dirk-thomas, 2016-10-24:
I am not sure if that paragraph is accurate in terms of what is desired. The API documentation for the `TransportHints` says something different: http://docs.ros.org/kinetic/api/roscpp/html/classros_1_1TransportHints.html#details

So currently if you want to prefer an unreliable transport but allow a reliable transport to be picked you need to specify both in that order: `TransportHints().unreliable().reliable()`

I would lean towards updating the referenced paragraph in the wiki.

guillaumeautran, 2016-10-24:
This is pretty misleading to call these 'hints' if they really are not hints. I say they are not hints mostly because if specified, the code will use them verbatim (as opposed to preference based). 
That's ok with me though but I find it confusing...

dirk-thomas, 2016-10-24:
Well, they can still work as "hints". But you basically have to set all allowed transports in the desired order. Then they will be considered.

On the other hand if you know that you don't want to use `udp` and only specify a `tcp` hint you certainly don't want it to fall back to `udp`. So I think the need to specify all "valid" hints is reasonable.

I updated the paragraph in the wiki. I hope that clarifies the current behavior: http://wiki.ros.org/action/diff/roscpp/Overview/Publishers%20and%20Subscribers?action=diff&rev1=17&rev2=18

guillaumeautran, 2016-10-24:
Thanks! That works for me. Should I close this issue or should you?

closed 2016-10-24 by dirk-thomas