ros/ros_comm #907: Why does roscore/rosmaster not respect the ROS_MASTER_URI environment variable?

jdyb, 2016-09-28:
Hi!

As stated in the header, when I change ROS_MASTER_URI to contain a non-default port and then start
a roscore it does not use the provided port.

Specifically the roscore prints these lines (irrelevant lines cut for brevity):
```
WARNING: ROS_MASTER_URI port [11312] does not match this roscore [11311]
ROS_MASTER_URI=http://localhost:11311/
```

Now, I am aware of the existence of the -p flag of roscore, so there is no real problem for my current use case.
But I am somewhat confused as to why it is implemented this way?
All the other tools use the port from ROS_MASTER_URI, so why does the roscore not do this?

(I am currently using indigo, just in case it is due to the old version.)

dirk-thomas, 2016-09-28:
The environment variable is documented in the wiki http://wiki.ros.org/ROS/EnvironmentVariables#ROS_MASTER_URI

> `ROS_MASTER_URI` is a required setting that tells nodes where they can locate the master.

It does not affect how you start a standalone `roscore`. It is your choice to pass custom port numbers. If you start a launch file though it will invoke roscore (if not already running) with the port number from the environment variable.

I understand your expectation but can't provide any information why this was decided that way. But since a lot of code might rely on the current behavior it is unlikely to be changed.

jdyb, 2016-10-01:
> ```ROS_MASTER_URI``` is a required setting that tells nodes where they can locate the master.

Ah, I guess I made an incorrect assumption here, since the master is not usually considered being a node.

> I understand your expectation but can't provide any information why this was decided that way.

Fair enough.

>  But since a lot of code might rely on the current behavior it is unlikely to be changed.

Oh I do not expect it to change. Backwards compatibility is important.

Thank you for your quick answer!

closed 2016-10-01 by jdyb