ros/ros_comm #908: rostopic echo topic/field no longer works

mikepurvis, 2016-09-29:
Looks like this might be more fallout from e0f63978fe0dd239ba8002a7e79756cdf1a335d0:

```
$ rostopic echo /rosout/msg -n 1
Traceback (most recent call last):
  File "/opt/clearpath/2.2pre/autonomy/lib/python2.7/dist-packages/rostopic/__init__.py", line 940, in callback
    self.suffix + '\n')
  File "/opt/clearpath/2.2pre/autonomy/lib/python2.7/dist-packages/rostopic/__init__.py", line 876, in custom_strify_message
    val = value_transform(val)
  File "/opt/clearpath/2.2pre/autonomy/lib/python2.7/dist-packages/rostopic/__init__.py", line 1393, in value_transform
    class TransformedMessage(genpy.Message):
  File "/opt/clearpath/2.2pre/autonomy/lib/python2.7/dist-packages/rostopic/__init__.py", line 1396, in TransformedMessage
    __slots__ = val.__slots__[:]
AttributeError: 'str' object has no attribute '__slots__'

$ rostopic echo /rosout/topics -n 1
Traceback (most recent call last):
  File "/opt/clearpath/2.2pre/autonomy/lib/python2.7/dist-packages/rostopic/__init__.py", line 940, in callback
    self.suffix + '\n')
  File "/opt/clearpath/2.2pre/autonomy/lib/python2.7/dist-packages/rostopic/__init__.py", line 876, in custom_strify_message
    val = value_transform(val)
  File "/opt/clearpath/2.2pre/autonomy/lib/python2.7/dist-packages/rostopic/__init__.py", line 1393, in value_transform
    class TransformedMessage(genpy.Message):
  File "/opt/clearpath/2.2pre/autonomy/lib/python2.7/dist-packages/rostopic/__init__.py", line 1396, in TransformedMessage
    __slots__ = val.__slots__[:]
AttributeError: 'list' object has no attribute '__slots__'
```

@wkentaro @dirk-thomas Thoughts on a fix?

dirk-thomas, 2016-09-29:
Thank you for reporting this. I can confirm that this is a regression :disappointed: I am looking into it since we really want to sync Kinetic before ROSCon...

wkentaro, 2016-09-29:
Sorry that I could not catching the case, #909 will fix the problem.

```
% rostopic echo /rosout/msg -n1
hello world 1475185572.54
---

% rostopic echo /rosout/topics -n1
['/chatter', '/rosout']
---

% rostopic echo /rosout/topics -n1 --noarr
None
---

% rostopic echo /rosout/msg -n1 --nostr
None
---
```

closed 2016-09-30 by dirk-thomas