ros/ros_comm #913: tf::MessageFilter CPU usage / clean up problem

conqualifaction, 2016-10-13:
The following code creates additional CPU usage if "testMethod" is called. The problem seems to be connected to the creation of MessageFilter. This can be tested by removing the MessageFilter creation code line.

The additional CPU usage is 4 % on my system.
I could not find a way to clean up in a way to avoid this (e.g. call clear(), unsubscribe(), etc. have no effect).

Tested on ROS Indigo.

```cpp
#include <ros/ros.h>
#include <image_transport/image_transport.h>
#include <tf/transform_listener.h>
#include <tf/message_filter.h>
#include <message_filters/subscriber.h>

void testMethod(ros::NodeHandle node)
{
  tf::TransformListener tf_listener(node, ros::Duration(40));
  message_filters::Subscriber<sensor_msgs::Image> img_subscriber(node, "/topic", 400);
  tf::MessageFilter<sensor_msgs::Image> msg_filter(img_subscriber, tf_listener, "/frame", 400);
}

int main(int argc, char** argv)
{
  ros::init(argc, argv, "test_node");
  ros::NodeHandle node;

  testMethod(node);

  ROS_INFO("Going to spin...");
  ros::spin();
  return 0;
}
```

dirk-thomas, 2016-10-14:
Your descriptions seems to indicate a problem with `tf::MessageFilter`. Please consider filling an issue in the repository which contains that class: https://github.com/ros/geometry/blob/indigo-devel/tf/include/tf/message_filter.h

conqualifaction, 2016-10-14:
Done. [https://github.com/ros/geometry/issues/132](url)

closed 2016-10-14 by conqualifaction