ros/ros_comm #915: rostopic echo -p no longer works

geoffviola, 2016-10-21:
There seems to be a new bug with rostopic echo, when printing in the csv format.

It looks like the call at https://github.com/ros/ros_comm/blob/kinetic-devel/tools/rostopic/src/rostopic/__init__.py#L936 expects value_transform as the parameter name. Meanwhile, the function defined at https://github.com/ros/ros_comm/blob/kinetic-devel/tools/rostopic/src/rostopic/__init__.py#L725 takes value_transform_fn as a parameter. Changing either seems to work as long as they're consistent, but I'm not sure which is better to change.

wjwwood, 2016-10-21:
Is this related to https://github.com/ros/ros_comm/pull/909?

geoffviola, 2016-10-21:
I don't think so. This error is specific to printing in the CSV format. Printing normally works fine.

geoffviola, 2016-10-21:
It looks related to this commit: https://github.com/ros/ros_comm/commit/e0f63978fe0dd239ba8002a7e79756cdf1a335d0

dirk-thomas, 2016-10-21:
@wkentaro can you please check if this is a regression and if yes work on a fix?

wkentaro, 2016-10-21:
@geoffviola Could you give me commands how to reproduce the error?

geoffviola, 2016-10-21:
### Window 1

1. source /opt/ros/kinetic/setup.bash
2. roscore

---

### Window 2

1. source /opt/ros/kinetic/setup.bash
2. rosrun roscpp_tutorials talker

---

### Window 3

1. source /opt/ros/kinetic/setup.bash
2. rostopic echo /chatter -p

PR #916 (2016-10-21): [rostopic] Fix typo of arg for _str_plot function
Fix #915
commit 010c367d (2016-10-21): Fix typo of arg for _str_plot function

Fix #915
closed 2016-10-21 by dirk-thomas