ros/ros_comm #923: roslz4 linking against the wrong python version

spaghetti-, 2016-10-31:
Something with the CMakeFiles.txt of roslz4 makes it disregard python overrides. On my most recent compilation os MacOS Sierra -

```
./src/catkin/bin/catkin_make_isolated --install --install-space /opt/ros/kinetic -DSETUPTOOLS_DEB_LAYOUT=OFF -DCMAKE_BUILD_TYPE=RelWithDebInfo  -DCMAKE_PYTHON_EXECTUABLE=`which python` -DCMAKE_PYTHON_LIBRARY=/usr/local/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib --pkg roslz4
```

```
❯ python -c "import rosbag"
Fatal Python error: PyThreadState_Get: no current thread
[1]    89264 abort      python -c "import rosbag"
```

Because

```
❯ otool -L _roslz4.so
_roslz4.so:
        _roslz4.so (compatibility version 0.0.0, current version 0.0.0)
        libroslz4.dylib (compatibility version 0.0.0, current version 0.0.0)
        /System/Library/Frameworks/Python.framework/Versions/2.7/Python (compatibility version 2.7.0, current version 2.7.10)
        /usr/local/opt/lz4/lib/liblz4.1.dylib (compatibility version 1.0.0, current version 1.7.1)
        /usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1238.0.0)

```

Fixed by 

```
install_name_tool -change /System/Library/Frameworks/Python.framework/Versions/2.7/Python /usr/local/Frameworks/Python.framework/Versions/2.7/Python _roslz4.so
```

It shouldn't happen because I specified my python libs, so it could be a bug.

dirk-thomas, 2016-10-31:
The package is using the [Python include dir](https://github.com/ros/ros_comm/blob/753ff953c689b3e0f4078b50a0193e04e36279a5/utilities/roslz4/CMakeLists.txt#L40) as well as [Python library](https://github.com/ros/ros_comm/blob/753ff953c689b3e0f4078b50a0193e04e36279a5/utilities/roslz4/CMakeLists.txt#L49) detected in CMake (https://github.com/ros/ros_comm/blob/753ff953c689b3e0f4078b50a0193e04e36279a5/utilities/roslz4/CMakeLists.txt#L39). I don't know if / how the CMake module takes your arguments into considerations. The code of the Python CMake module will show what is happening and if there is a way to pass a specific library dir.

mikepurvis, 2017-02-09:
In **ros-install-osx**, we set `PYTHON_LIBRARY` and `PYTHON_INCLUDE_DIR`; this has seemed to work consistently, at least when targeting brewed python:

https://github.com/mikepurvis/ros-install-osx/blob/fb9febdcfa9b965b0b6f347f4963d9277b1d1b26/install#L159-L160

mikepurvis, 2017-02-14:
Note that as of Trusty, there's an upstream `lz4` package; as of Xenial, there's an upstream `python-lz4` bindings package:

http://packages.ubuntu.com/source/trusty/lz4
http://packages.ubuntu.com/source/xenial/python-lz4

Potentially for Lunar+, the roslz4 package could be hollowed out and made to delegate to these system implementations.

oliverlee, 2018-04-18:
> In ros-install-osx, we set PYTHON_LIBRARY and PYTHON_INCLUDE_DIR; this has seemed to work consistently, at least when targeting brewed python:

This still links to the wrong Python (at least for me) since roslz4 uses (deprecated) `PYTHON_INCLUDE_PATH` instead of `PYTHON_INCLUDE_DIR`.

alper111, 2020-01-11:
This is still an issue on MacOS Catalina, and the solution @spaghetti- mentioned works for me.

commit eaa96d1f (2020-04-20): Fix #923 (use undefined dynamic_lookup on OSX) (#1923)

See https://github.com/conda-forge/ros-roslz4-feedstock/pull/3 and https://gitter.im/RoboStack/Lobby?at=5e830ffdbd09254f83ea7d4c and https://blog.tim-smith.us/2015/09/python-extension-modules-os-x/ for details.
commit 91427bd8 (2020-05-29): Fix #923 (use undefined dynamic_lookup on OSX) (#1923)

See https://github.com/conda-forge/ros-roslz4-feedstock/pull/3 and https://gitter.im/RoboStack/Lobby?at=5e830ffdbd09254f83ea7d4c and https://blog.tim-smith.us/2015/09/python-extension-modules-os-x/ for details.
closed 2020-04-20 by dirk-thomas