ros/ros_comm #935: rostest result xml always reports failures=0 even when there are failures

nwang2, 2016-11-16:
Attached is the console output:
[f200-disable-color.txt](https://github.com/ros/ros_comm/files/593600/f200-disable-color.txt)
But the test result xml file is as below:
<?xml version="1.0" encoding="UTF-8"?>
-<testsuite time="33.994" tests="1" name="unittest.suite.TestSuite" failures="0" errors="0">
<testcase time="33.9936" name="testrealsense_camera_test" classname="rostest.runner.RosTest"/>

dirk-thomas, 2016-11-16:
The console output alone doesn't say much. Can you please provide a sequence of commands to reproduce the problem.

nwang2, 2016-11-16:
Running command:
rostest realsense_camera f200_nodelet_disable_color.test
We see failures in console output like I attached above. But the xml file reports "failures=0".

dirk-thomas, 2016-11-16:
Please provide enough information to reproduce the problem. Which ROS distro are you using? Which repos are you trying to build / test?

nwang2, 2016-11-16:
This issue can be reproduced on both indigo and kinetic.
I am building and testing https://github.com/intel-ros/realsense/tree/indigo-devel/realsense_camera.
When running unit tests by rostest with test files from https://github.com/intel-ros/realsense/tree/indigo-devel/realsense_camera/test, I can see this problem.