ros/ros_comm #940: Race condition in rosunit.core

MarcSeebold, 2016-11-25:
Hi there,
we experienced a race condition when running unit tests using rosunit.

Version: ros-jade-rosunit 1.12.7-0trusty-20160422-184406-0700 amd64

`[ROSUNIT] Outputting test results to /home/jenkins/workspace/repository__clang__nightly/build/some_package/test_results/some_package/rostest-test_reporter_test.xml
Traceback (most recent call last):
Traceback (most recent call last):
  File "/opt/ros/jade/share/rostest/cmake/../../../bin/rostest", line 36, in <module>
  File "/opt/ros/jade/share/rostest/cmake/../../../bin/rostest", line 36, in <module>
        rostestmain()
rostestmain()
  File "/opt/ros/jade/lib/python2.7/dist-packages/rostest/__init__.py", line 273, in rostestmain
  File "/opt/ros/jade/lib/python2.7/dist-packages/rostest/__init__.py", line 273, in rostestmain
    _main()
  File "/opt/ros/jade/lib/python2.7/dist-packages/rostest/rostest_main.py", line 167, in rostestmain
    results_file = xmlResultsFile(pkg, outname, is_rostest, env=env)
  File "/opt/ros/jade/lib/python2.7/dist-packages/rosunit/core.py", line 102, in xml_results_file
    _main()
  File "/opt/ros/jade/lib/python2.7/dist-packages/rostest/rostest_main.py", line 167, in rostestmain
    results_file = xmlResultsFile(pkg, outname, is_rostest, env=env)
  File "/opt/ros/jade/lib/python2.7/dist-packages/rosunit/core.py", line 102, in xml_results_file
    raise IOError("cannot create test results directory [%s]. Please check permissions."%(test_dir))
IOError: cannot create test results directory [/home/jenkins/workspace/pjfa_repository__clang__nightly/build/some_package/test_results/some_package]. Please check permissions.`

Running the same configuration over-and-over we noticed that it is non-deterministic (more or less 1 out of 10 fails).

Looking in the corresponding code in rosunit/core.py shows:

`86 def xml_results_file(test_pkg, test_name, is_rostest=False, env=None): 
 87      """ 
 88      @param test_pkg: name of test's package  
 89      @type  test_pkg: str 
 90      @param test_name str: name of test 
 91      @type  test_name: str 
 92      @param is_rostest: True if the results file is for a rostest-generated unit instance 
 93      @type  is_rostest: bool 
 94      @return: name of xml results file for specified test 
 95      @rtype:  str 
 96      """ 
 97      test_dir = os.path.join(rospkg.get_test_results_dir(env=env), test_pkg) 
 98      if not os.path.exists(test_dir): 
 99          try: 
100              makedirs_with_parent_perms(test_dir) 
101          except OSError: 
102              raise IOError("cannot create test results directory [%s]. Please check permissions."%(test_dir)) `

Un-commenting _except OSError_ reveals the true error message:
`[ROSUNIT] Outputting test results to /home/sem2pal/workspace/build/some_package/test_results/some_package/rostest-test_requester_test.xml
Traceback (most recent call last):
  File "/opt/ros/jade/share/rostest/cmake/../../../bin/rostest", line 36, in <module>
    rostestmain()
  File "/opt/ros/jade/lib/python2.7/dist-packages/rostest/__init__.py", line 273, in rostestmain
    _main()
  File "/opt/ros/jade/lib/python2.7/dist-packages/rostest/rostest_main.py", line 167, in rostestmain
    results_file = xmlResultsFile(pkg, outname, is_rostest, env=env)
  File "/opt/ros/jade/lib/python2.7/dist-packages/rosunit/core.py", line 100, in xml_results_file
    makedirs_with_parent_perms(test_dir)
  File "/opt/ros/jade/lib/python2.7/dist-packages/rosunit/core.py", line 77, in makedirs_with_parent_perms
    os.mkdir(p)
OSError: [Errno 17] File exists: '/home/sem2pal/workspace/build/some_package/test_results/some_package'`

So... it seems the _if not os.path.exists(test_dir)_ failed.
Under the assumption that there are multiple tests executed in parallel the race condition is obvious.

As a possible fix I recommend to add error handling for errno.EEXIST:
`except OSError as e:
    if e.errno == errno.EEXIST:
        pass
    raise IOError("cannot create test results directory [%s]. Please check permissions."%(test_dir))`
for the code in
def create_xml_runner(test_pkg, test_name, results_file=None, is_rostest=False)
and
def xml_results_file(test_pkg, test_name, is_rostest=False, env=None):

dirk-thomas, 2016-11-25:
Thank you for reporting this. The problem seems to be in a different repo. Can you please try the proposed fix from ros/ros#130. The patch looks a bit different to what you described since it needs to take the permission into account.

MarcSeebold, 2016-11-25:
@dirk-thomas That fixes it. Thx!

closed 2016-11-29 by dirk-thomas