ros/ros_comm #944: [rosbag] speedier lookups with "rosbag info -y --key"

jacquelinekay, 2016-11-30:
`rosbag info -y --key <filename>` offers a nice command line syntax for querying rosbag metadata such as size and duration. However, for very large, uncompressed, indexed bags, it can take a very long time to access these attributes (very long = over 10 seconds). It would be great if we only had to pay in time for the information we want to retrieve.

I haven't benchmarked to see where the bottleneck is. It's probably in the constructor of Bag, and a speedup might be possible from dissecting `Bag.read_file_header_record()` a bit.

spaghetti-, 2016-12-28:
How big does the bagfile have to be before this becomes noticeable?

NikolausDemmel, 2018-03-05:
I imagine that this relates to long bag opening times with large number of chunks and can possibly be worked around by using a large chunksize, see https://github.com/ros/ros_comm/issues/117.

closed 2025-05-31 by sloretz