ros/ros_comm #946: Create public Python function for rostopic list in rostopic package

gordonbrander, 2016-12-06:
Currently, the only function `rostopic` has for listing topics [is private](https://github.com/ros/ros_comm/blob/kinetic-devel/tools/rostopic/src/rostopic/__init__.py#L1181), and it is only useful for thecli usecase because it uses `print`.

`rosgraph.Master("/rostopic").getSystemState()` may be used instead of rostopic, but this seems like an abuse of an API which is meant for graph image generation.

I would like to propose that a public `rostopic.list` function be added to the `rostopic` package, listing all topic names (and perhaps types).

mikepurvis, 2017-02-14:
I don't disagree with you, but FWIW, what you suggest is exactly how rosbridge handles inspecting the system:

https://github.com/RobotWebTools/rosbridge_suite/blob/a7a2e1a473a1da1acf252d2496fa7db0f211a29f/rosapi/src/rosapi/proxy.py#L49-L56

zwn, 2020-10-22:
Should this be closed since get_topic_list() has been merged?

closed 2025-05-31 by sloretz