ros/ros_comm #953: roslaunch_add_file_check doesn't support args

damonkohler, 2016-12-22:
We have some top-level launch files that expect args to be passed in on the command line. As far as I can tell, roslaunch_add_file_check only supports setting environment variables. So, the test fails when we have an undefined arg in a top-level launch file.

dirk-thomas, 2016-12-22:
Can you please clarify how it fails in your case. Maybe it is similar to other pending tickets.

damonkohler, 2017-01-05:
I'm not sure exactly what further clarification you're looking for. I've pasted a failing log snippet below though in case that helps.

```
cd /usr/local/google/home/damonkohler/catkin_ws/build_isolated/cartographer_ros && /usr/local/google/home/damonkohler/catkin_ws/build.../cartographer_ros/roslaunch-check_launch.xml'\ '/usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch'\
FAILURE:
[/usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/offline_backpack_2d.launch]:
        [/usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/offline_backpack_2d.launch] requires the 'bag_filenames' arg to be set
[/usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/demo_backpack_3d.launch]:
        [/usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/demo_backpack_3d.launch] requires the 'bag_filename' arg to be set
[/usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/demo_backpack_2d.launch]:
        [/usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/demo_backpack_2d.launch] requires the 'bag_filename' arg to be set
[/usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/demo_pr2.launch]:
        [/usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/demo_pr2.launch] requires the 'bag_filename' arg to be set
[/usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/demo_taurob_tracker.launch]:
        [/usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/demo_taurob_tracker.launch] requires the 'bag_filename' arg to be set
[/usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/offline_backpack_3d.launch]:
        [/usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/offline_backpack_3d.launch] requires the 'bag_filenames' arg to be set
[/usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/demo_revo_lds.launch]:
        [/usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/demo_revo_lds.launch] requires the 'bag_filename' arg to be set
checking *.launch in directory /usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch
checking /usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/offline_backpack_2d.launch
checking /usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/demo_backpack_3d.launch
checking /usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/demo_backpack_2d.launch
checking /usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/demo_pr2.launch
checking /usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/demo_taurob_tracker.launch
checking /usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/offline_backpack_3d.launch
checking /usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/backpack_3d.launch
checking /usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/taurob_tracker.launch
checking /usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/backpack_2d.launch
checking /usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer_ros/launch/demo_revo_lds.launch
...writing test results to /usr/local/google/home/damonkohler/catkin_ws/build_isolated/cartographer_ros/test_results/cartographer_ros/roslaunch-check_launch.xml
wrote test file to [/usr/local/google/home/damonkohler/catkin_ws/build_isolated/cartographer_ros/test_results/cartographer_ros/roslaunch-check_launch.xml]
-- run_tests.py: execute commands
  /usr/bin/cmake -E make_directory /usr/local/google/home/damonkohler/catkin_ws/build_isolated/cartographer_ros/test_results/cartographer_ros
  /opt/ros/indigo/share/roslaunch/cmake/../scripts/roslaunch-check -o /usr/local/google/home/damonkohler/catkin_ws/build_isolated/cartographer_ros/test_results/cartographer_ros/roslaunch-check_launch.xml /usr/local/google/home/damonkohler/catkin_ws/src/cartographer_ros/cartographer
_ros/launch
-- run_tests.py: verify result "/usr/local/google/home/damonkohler/catkin_ws/build_isolated/cartographer_ros/test_results/cartographer_ros/roslaunch-check_launch.xml"
```

mikepurvis, 2017-02-09:
We have this issue too, and work around it by always ensuring that top level launch files have default values for their arguments, and included launch files which require them are only ever tested via inclusion, rather than directly.

shaun-edwards, 2017-05-19:
We solved a similar problem in ROS-I by creating test specific launch [files](https://github.com/ros-industrial/motoman/blob/jade-devel/motoman_mh5_support/test/launch_test.xml).  In our case, we didn't feel comfortable with default values because there wasn't a valid default or the default could lead an unexpected/dangerous situation.

130s, 2017-12-06:
Forgive me if I'm misunderstanding the issue discussed here, but the title sounds like an exact issue I have.

`roslaunch_add_file_check` didn't pick up a missing dependency (say X) where X appears only in `args` in `node` element for launching `Nodelet`.

Example ([openni2_launch](https://github.com/ros-drivers/openni2_camera/blob/536bb2819c7a4b06ff53e2c07eff0c0f7b68089c/openni2_launch/launch/includes/device.launch.xml#L34)):
```
  <node pkg="nodelet" type="nodelet" name="driver"
        args="load openni2_camera/OpenNI2DriverNodelet $(arg manager) $(arg bond)"
respawn="$(arg respawn)">
```
In this case `openni2_camera` in `args` won't be captured by roslaunch_add_file_check as a dependency (for this particular pkg openn2_launch, the dependency is defined so no problem. I'm using this portion only as a sample).

mintar, 2018-02-23:
@130s : No, your issue is different. Your issue is caused by the fact that openni2_camera is a command line argument to the executable `nodelet`, and roslaunch_add_file_check doesn't understand that `nodelet` expects a package name as part of the second command line argument string. This is not a bug in roslaunch_add_file_check.

This issue is that if you have a launch file like this:

```xml
<launch>
  <arg name="foo" />    <!-- no `default="something" here! -->
  <node name="bar" pkg="bar" type="bar">
    <param name="foo" value="$(arg foo)" />
  </node>
</launch>

```

... then roslaunch_add_file_check() will print:

```
FAILURE:
[/home/martin/ros/kinetic/ros_comm_953/src/roslaunch_test/launch/foobar.launch]:
	[/home/martin/ros/kinetic/ros_comm_953/src/roslaunch_test/launch/foobar.launch] requires the 'foo' arg to be set
```

130s, 2019-07-23:
My team is interested in providing a PR to address this.

Thanks @mintar for clarification https://github.com/ros/ros_comm/issues/953#issuecomment-367952313. To provide a situation where this issue can be an issue, in product development where the same rendition of the code gets installed on multiple instances of hosts, `default` value for some `arg`s don't often make sense (I think Shaun's case https://github.com/ros/ros_comm/issues/953#issuecomment-302796884 is similar / same), e.g. serial number of sensors and such, so keeping those blank and make them "requirement" for the downstream launch files is found to be a valid usecase (I'm open for better implementation). 

As for solution, making a check for `arg` optional might be a good start?

CC @collin-scribner

130s, 2020-04-27:
https://github.com/ros/ros_comm/pull/1788 is merged into `melodic-devel`. With that we may want to close this ticket.

closed 2020-05-13 by dirk-thomas

mintar, 2023-01-31:
Just for people finding this in the future:

#1788 doesn't always work, for example when the argument is used as `ns` in a `group` tag. In that case, you can set the argument like this:

On the command line:

```
rosrun roslaunch roslaunch-check launch/foobar.launch foo:=baz
```

In the CMakeLists.txt:

```
roslaunch_add_file_check(launch/foobar.launch foo:=baz)
```