ros/ros_comm #967: rosbag info doesn't show some topics

ojura, 2017-01-27:
I am using Kinetic on Ubuntu 16.04. At first I thought that I was going crazy, but `rosbag info` occasionally does not print all topics and it seems that it omits the last  - usually on the first call. Subsequently it prints everything correctly.

Here is an example:

```
juraj@juraj-laptop$ rosbag info amacal_2017-01-03-17-28-04-light-change.bag
path:        amacal_2017-01-03-17-28-04-light-change.bag
version:     2.0
duration:    3:48s (228s)
start:       Jan 03 2017 17:28:06.19 (1483460886.19)
end:         Jan 03 2017 17:31:54.46 (1483461114.46)
size:        5.3 GB
messages:    107756
compression: none [4517/4517 chunks]
types:       sensor_msgs/CameraInfo [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/Image      [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu        [6a62c6daae103f4ff57a132d6f95cec2]
             sensor_msgs/LaserScan  [90c7ef2dc6895d81024acba2ac42f369]
             timesync/TimesyncDebug [1b9fcff64ffd5b3203e0d1026359e2c2]
topics:      t20/camera/camera_info                          4514 msgs    : sensor_msgs/CameraInfo
             t20/camera/image_raw                            4516 msgs    : sensor_msgs/Image     
             t20/camera/timestamp_synchronizer/debug_info    4518 msgs    : timesync/TimesyncDebug
             t20/imu                                        45659 msgs    : sensor_msgs/Imu       
             t20/imu/timestamp_synchronizer/debug_info      45659 msgs    : timesync/TimesyncDebug

juraj@juraj-laptop$ rosbag info amacal_2017-01-03-17-28-04-light-change.bag
path:        amacal_2017-01-03-17-28-04-light-change.bag
version:     2.0
duration:    3:48s (228s)
start:       Jan 03 2017 17:28:06.19 (1483460886.19)
end:         Jan 03 2017 17:31:54.46 (1483461114.46)
size:        5.3 GB
messages:    107756
compression: none [4517/4517 chunks]
types:       sensor_msgs/CameraInfo [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/Image      [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu        [6a62c6daae103f4ff57a132d6f95cec2]
             sensor_msgs/LaserScan  [90c7ef2dc6895d81024acba2ac42f369]
             timesync/TimesyncDebug [1b9fcff64ffd5b3203e0d1026359e2c2]
topics:      t20/camera/camera_info                          4514 msgs    : sensor_msgs/CameraInfo
             t20/camera/image_raw                            4516 msgs    : sensor_msgs/Image     
             t20/camera/timestamp_synchronizer/debug_info    4518 msgs    : timesync/TimesyncDebug
             t20/imu                                        45659 msgs    : sensor_msgs/Imu       
             t20/imu/timestamp_synchronizer/debug_info      45659 msgs    : timesync/TimesyncDebug
             t20/laserscan                                   2890 msgs    : sensor_msgs/LaserScan
```

Notice that it has not printed the `t20/laserscan` topic on the first invocation, although it correctly listed that `sensor_msgs/LaserScan` messages are contained in the bag (and this message type is only published on that topic).

dirk-thomas, 2017-01-27:
I can't reproduce the problem with bag files at hand. Can you please share your bagfile to reproduce the problem.

ojura, 2017-01-27:
The issue is kind of hard to reproduce, I haven't noticed it is related to a particular bag, and it occurs rarely. Reporting it was kind of a shot in the dark. If I figure out what could be the trigger, I will update the issue.

closed 2025-05-31 by sloretz