ros/ros_comm #972: Cannot set a param to None

asmodehn, 2017-02-06:
Code like this : 
```python
rospy.set_param('~test_none', None)
```
or that:
```python
rospy.set_param('~test_none_bis', {"some_structure" : None})
```
will fail.

The error is : 
```python
File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 504, in set_param
    _param_server[param_name] = param_value #MasterProxy does all the magic for us
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/msproxy.py", line 148, in __setitem__
    self.target.setParam(rospy.names.get_caller_id(), rospy.names.resolve_name(key), val)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1581, in __request
    allow_none=self.__allow_none)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1086, in dumps
    data = m.dumps(params)
  File "/usr/lib/python2.7/xmlrpclib.py", line 633, in dumps
    dump(v, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 655, in __dump
    f(self, value, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 736, in dump_struct
    dump(v, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 655, in __dump
    f(self, value, write)
  File "/usr/lib/python2.7/xmlrpclib.py", line 659, in dump_nil
    raise TypeError, "cannot marshal None unless allow_none is enabled"
TypeError: cannot marshal None unless allow_none is enabled
```

This makes it harder than necessary to store any kind of `dict` structure in params, as one needs to filter for `None`.

So I was wondering what was the reasoning behind not allowing `None` in params ?
If we can consider improving this, what could be the undesired side effects of setting allow_none for our xmlrpcapi ?

mikepurvis, 2017-02-09:
rosparams can be present or absent, but if present they are expected to comply to a type, otherwise trying to use them in C++ would be madness, for example:

```
std::string my_param;
ros::param::param<std::string>("param_name", my_param, "default_value");
```

If a parameter were able to exist but be `None`, what is the correct behaviour here? It would either have to return the default value, which isn't really correct, or do something totally new like return an error code or throw an exception, none of which are part of the current contract for this function.

asmodehn, 2017-02-09:
Ah I see. This is the same issue as not being able to have a None/null in a
message field or an optional field.

It comes from the fact that the data model and the API is designed using
direct memory structure, C style, instead of using handles (type storing a
pointer/reference to the data,like a smart pointer for example) which is
definitely easier to deal with in C++11 than C...

That makes the usage with higher level languages (python in our case) quite
frustrating in some places, as adding that indirection level later is a bit
harder...

Parameters map a python dictionnary nicely, so I was expecting to have a
dict like interface in C++ as well and be able to put empty handles in
there.

Anyway lets close that, as we probably wont change the API to use handles
now...

Thanks for the reply anyway.

<https://github.com/ros/ros_comm/issues/972#issuecomment-278662791>, or mute
the thread

closed 2017-02-09 by dirk-thomas