ros/ros_comm #974: ROS missing callbacks, or not calling at a fast enough rate

anuppari, 2017-02-07:
There is something wrong with how ROS is calling subscriber callbacks at a much slower rate than msgs are being published. Here's a minimal example.

```cpp
class DataHandler
{
private:
  ros::NodeHandle nh;
  ros::Publisher test_pub;
public:
  DataHandler() { test_pub = nh.advertise<geometry_msgs::PoseStamped>("test",10); }
  void callback(const gazebo_msgs::LinkStatesConstPtr& msg)
  {
    test_pub.publish(geometry_msgs::PoseStamped());
  }
};

int main(int argc, char** argv)
{
  ros::init(argc, argv, "gazebo_to_ros");
  ros::NodeHandle nh;

  DataHandler callbacks;
  ros::Subscriber linkSub = nh.subscribe("gazebo/link_states",1,&DataHandler::callback,&callbacks);

  ros::spin();
  return 0;
}
```

The topic "gazebo/link_states" is being published to at 500 Hz (using `rostopic hz` at the command line). The code in the callback takes a trivial amount of time to run (using [std::chrono](http://www.cplusplus.com/reference/chrono/), less than 1ms => can run >1000Hz). The topic "test" is only being published to at ~70 Hz. Time between callback calls fluctuates between 1e-5 s and 0.04s, => 100kHz to 25 Hz, yielding an average of ~70 Hz according to `rostopic hz`.

Resource monitor shows cpu usage at less than 50%. Even setting high priority for this process changes nothing, so I don't think its a bottleneck in the hardware.

I'm on Ubuntu 16.04 LTS, with ROS Kinetic. Anyone have any idea at what I should look at next to help troubleshoot? Thanks for your help.

dirk-thomas, 2017-02-07:
Have you tried to set a custom queue size for your publisher? It is very likely that the publishing thread can't keep up with the rate and drops messages when the queue limit is reached.

anuppari, 2017-02-07:
Do you mean the "gazebo/link_states" topic? That is published by gazebo, so I don't have control over it. However, I've run into this same issue before where I did set the corresponding queue size to ~10, and still had this issue.

The callback runs very fast, ~order of magnitude faster than messages are being published to the topic, so I don't see how it would be missing messages.

dirk-thomas, 2017-02-07:
Can you provide a small, self contained example to reproduce the problem (http://sscce.org/).

dirk-thomas, 2017-02-07:
I don't expect this to be a bug in ROS. Therefore [answers.ros.org](http://answers.ros.org/) might be a better place to raise your question though since many more people can provide feedback, suggestions, and answers.

anuppari, 2017-02-07:
I made a basic [talker/listener](https://github.com/anuppari/basic_package) to test this. If I run the pair of nodes with nothing else running, I can get 10kHz communication without message drop. I can even run 2 pairs of talkers/listeners at 10kHz fine. However, if I run my sim in parallel, with many high rate topics, I get significant message drop, despite everything communicating on separate topics. Is there a bottleneck somewhere? With everything running the CPU usage is still only around 50%.

Interestingly, when the sim is running in parallel, the talker rate also decreases to 1kHz, (and the listener callback only runs ~600Hz) despite significant cpu headroom.

I also don't think its related to Gazebo, mavros, px4 or any of the other nodes I'm running in my sim. I've run into this same issue before with my own simple sim, with many fewer topics, though still ~3 high rate (>200 Hz) topics.

Is there some kind of bottleneck in the communication layer, something related to how ROS uses sockets, tcp, etc? From my understanding, each publisher/subscriber pair is on a separate p2p channel, so I don't see why there would be this kind of bottleneck.

dirk-thomas, 2017-02-07:
I am not aware of any bottleneck like this in `roscpp`.

Regarding your CPU usage: could it be that the "50%" are actually fully loading some of your cores which are then limiting your performance? Maybe only other cores are idle but some of your processes are still CPU bound on a specific core.

anuppari, 2017-02-07:
No all cores were showing at ~50%. And even if that were the case, the cpu scheduler would put the talker/listener nodes on a separate core, right? Since the talker/listener are completely independent of everything else, as long as 1 core is not maxed out, they should be able to perform at the prescribed rate, correct?

dirk-thomas, 2017-02-07:
Yes, separate processes can be balances to separate cores.

I can't provide more feedback on this topic without a minimal reproducible example (http://sscce.org/). Please consider asking your question on http://answers.ros.org where it will be visible to more people. You can post the link to your question here for cross referencing.

anuppari, 2017-02-07:
Here's the question on ROS answers: http://answers.ros.org/question/253942/ros-not-calling-callbacks-fast-enough/

OK I've updated my [package](https://github.com/anuppari/basic_package) to reproduce the problem. Again, it's just a basic talker/listener, where the talker is sending msgs of type `gazebo_msgs::LinkStates` at 1kHz. I noticed that if there are less than 5 links in the message, the listener callback also runs at 1kHz. If there are 5 or more, the listener only runs at ~170Hz. The callback itself is trivial and shouldn't be a bottleneck.

With this example, all cores on my CPU are running below 20%. I'm running an Intel Xeon E5-1607v2 @ 3GHz in case it's relevant.

Could this be an issue with some packet size or buffer setting in the tcp/socket? Like the socket buffers a bunch of tcp packets and then sends them all in one go? If there is more data in the message than the buffer, the socket "blocks" until all the data in the buffer has been sent and the buffer is clear before allowing more data to be sent to the socket?

anuppari, 2017-02-07:
In case anyone arrives here through google, I tried the [noTcpDelay](http://wiki.ros.org/roscpp/Overview/Publishers%20and%20Subscribers#Transport_Hints) transport hint, and that seems to have fixed the issue.

Is this something worth making default for large message types? From some basic googleing, it seems like this is a common issue for large messages like Odometry. Since the subscriber knows the message type at initialization, should the noTcpDelay option automatically be set to true for message types that serialize to over a certain size? It might be difficult for variable length fields.

dirk-thomas, 2017-02-09:
I don't think that making the option the default is reasonable. Too much code already relies on the current behavior and changing that will likely break some of them. It might be good to improve documentation on this to make it easier to discover. If you have any specific suggestions please feel free to propose them here or update the wiki directly.

I will close this then since it is solved by passing `noTcpDelay`.

closed 2017-02-09 by dirk-thomas

dhiegomaga, 2019-07-08:
I had a similar issue, where the node was taking 5 ~ 7 sec between callbacks, given the publisher had a 20 Hz rate (it was odometry). It only happened when I ran other nodes that were CPU and GPU demanding, however I had a lot of CPU time still available. 

After trying using [noTcpDelay()](http://wiki.ros.org/roscpp/Overview/Publishers%20and%20Subscribers#Transport_Hints), increasing queue size, [bounding nodes to specific processor cores](https://answers.ros.org/question/202712/how-to-bind-a-node-to-a-specific-cpu-core/) and [changing node execution priority](https://answers.ros.org/question/246090/how-to-launch-nodes-with-realtime-priority/) without results. 

Finally I tried using different ros [Multi-threaded Spinning mechanism](http://wiki.ros.org/roscpp/Overview/Callbacks%20and%20Spinning#Multi-threaded_Spinning) and now it works faster than never. 

[This](https://answers.ros.org/question/53055/ros-callbacks-threads-and-spinning/) helped me arrive at a solution.

FahadRazaKhan, 2020-05-04:
I had the same problem while subscribing to Nav_Odometry messages and it has been solved by simply tcpNoDelay. Thank you.