ros/ros_comm #987: rosbag: Start/End time differ by one second for same timestamp

felixendres, 2017-02-14:
After filterering two bagfiles to remove the overlap, I have the following output of rosbag info

    path:        AET-raw_2017-02-07-10-21-25-filtered.bag
    version:     2.0
    duration:    44:57s (2697s)
    start:       Feb 07 2017 10:21:25.62 (1486459285.62)
    end:         Feb 07 2017 11:06:22.00 (1486461983.00)

and

    path:        AET-raw_2017-02-07-11-06-23_0-filtered.bag
    version:     2.0
    duration:    29:07s (1747s)
    start:       Feb 07 2017 11:06:23.00 (1486461983.00)
    end:         Feb 07 2017 11:35:30.56 (1486463730.56)

Note how the same timestamp (1486461983.00) shows up as 11:06:22 and 11:06:23. According to `date --date='@1486461983'` the second variant is correct.

The filtering commands I used were
`rosbag filter AET-raw_2017-02-07-10-21-25.bag AET-raw_2017-02-07-10-21-25-filtered.bag "t.secs < 1486461983.0"`
`rosbag filter AET-raw_2017-02-07-11-06-22_0.bag AET-raw_2017-02-07-11-06-23_0-filtered.bag "t.secs >= 1486461983.0"`

felixendres, 2017-02-14:
Clicking through the messages in rqt_bag, I found messages at the end of the first file that are timestamped later than 1486461983, the maximum stamp in my filter command. I.e. it seems that filtering didn't work right.

However, stepping backwards from the end (timestamp still being > 1486461983) I get the following error in the terminal where rqt_bag is running:

> Exception in thread Thread-10:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/message_loader_thread.py", line 83, in run
    msg_data = self._get_message(bag, playhead_position)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/message_loader_thread.py", line 96, in _get_message
    msg_data = self.timeline.read_message(bag, position)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_bag/bag_timeline.py", line 480, in read_message
    return bag._read_message(position)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosbag/bag.py", line 980, in _read_message
    return self._reader.seek_and_read_message_data_record(position, raw)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosbag/bag.py", line 2474, in seek_and_read_message_data_record
    data = _read_record_data(f)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosbag/bag.py", line 1711, in _read_record_data
    raise ROSBagFormatException('Error reading record data: %s' % str(ex))
ROSBagFormatException: Error reading record data: error unpacking uint32: unpack requires a string argument of length 4

I am quite confused now as to where the error lies.

felixendres, 2017-02-14:
> Clicking through the messages in rqt_bag, I found messages at the end of the first file that are timestamped later than 1486461983, the maximum stamp in my filter command. I.e. it seems that filtering didn't work right.

Please ignore the above paragraph, I confused timestamps *in* the message with timestamps *of* the message, i.e. the timestamp contained in the header vs the time when the message was received by rosbag record (running on another computer).

dirk-thomas, 2017-07-17:
If you run `rosbag filter` with the verbose option `-p "'%d.%d' % (t.secs, t.nsecs)"` it will output a message for every match / not matched timestamp.

The expression you are using is incorrect: `"t.secs < 1486461983.0"`. `tsecs` is just an integer and can't be compared to a floating point number. Therefore the one second rounding will occur.

Please see the [wiki page](http://wiki.ros.org/rosbag/Commandline#filter) which provides an example how this can be used: `"t.to_sec() <= 1284703931.86"`. That should give you the exact result your are expecting.

I will close the ticket for now since I don't think it is a bug but a usage issue. Please feel free to comment and the ticket can be reopened if necessary.

For future questions like this I would recommend to ask on answers.ros.org first since you would very likely have received more timely support from other users in the community.

closed 2017-07-17 by dirk-thomas