ros/ros_comm #993: [roslaunch] roslaunch-check doesn't respect if/unless

mikepurvis, 2017-02-18:
We're having a case where we have some launch files which optionally include files based on an environment variable being set. These are failing the test created by `roslaunch_add_file_check`, since it doesn't respect the conditional and attempts to resolve all include elements without regard for context.

This can be reproduced trivially by a launch file such as this, which will (wrongly, in my opinion) fail the roslaunch check:

```
<launch>
    <include if="0" file="does/not/exist.launch" />
</launch>
```

mikepurvis, 2017-02-21:
I believe the issue is here:

https://github.com/ros/ros_comm/blob/674263142a0be1124b273bdb1e0a79c3c956e259/tools/roslaunch/src/roslaunch/depends.py#L144-L148

It needs logic to explicitly check for if/unless, similar to how the args work:

https://github.com/ros/ros_comm/blob/674263142a0be1124b273bdb1e0a79c3c956e259/tools/roslaunch/src/roslaunch/depends.py#L99-L106

What's mysterious to me is that this works correctly for a normal launch but not for `roslaunch-check`. Is there some preprocessing going on that strips out tags based on if/unless attributes?

dirk-thomas, 2017-04-28:
Addressed by #998. Thank you!

closed 2017-04-28 by dirk-thomas