ros/ros_comm #995: Passing sigstop / sigcont

el3ment, 2017-02-21:
Currently roslaunch passes the sigint signal through to the child processes, but does not pass any of the other signals through. Ideally I would have been able to send a sigstop command to roslaunch, and pause all the subprocesses as well, and then continue all of them with a sigcont.

dirk-thomas, 2017-02-21:
While this is an interesting feature the maintainers will likely not have time to implement it. Please consider to provide a PR for this (against the default branch of the repo).

LightnessOfBeing, 2017-04-25:
@el3ment Did you mean processes that were launched on remote machine?

el3ment, 2017-04-25:
It is also true for remote processes, but this issue was originally referring to normal processes.

Robert

evenator, 2017-05-26:
If implemented, I would like this to affect both local and remote processes. Stopping some but not all processes with sigstop would be worse than current behavior for me.

closed 2025-05-31 by sloretz