ros/ros_comm #1007: wiki documentation for "unless" is confusing

goretkin, 2017-03-01:
(If there is a more appropriate place to report issues with the wiki content, please let me know)

The wording at http://wiki.ros.org/roslaunch/XML#if_and_unless_attributes

> if=value (optional)
>    If value evaluates to true, include tag and its contents. 

> unless=value (optional)
>    Unless value evaluates to false, include tag and its contents.

is confusing (and hopefully wrong) since in my mind "if a" is equivalent to "unless not a".

The wording in the subsequent example is clear.

dirk-thomas, 2017-03-01:
Thank for reporting. It looks like someone change the correct content: http://wiki.ros.org/action/diff/roslaunch/XML?action=diff&rev1=40&rev2=41

I fixed it in http://wiki.ros.org/action/diff/roslaunch/XML?action=diff&rev1=46&rev2=47 Please feel free to edit the wiki page directly in the future if you are pretty sure about the change.

closed 2017-03-01 by dirk-thomas

130s, 2017-03-01:
Apologies it was me who added the wrong change. 
Latest change Dirk made is even clearer than ever to me. Thanks!