ros/ros_comm #1019: Duration bug

asmodehn, 2017-03-17:
While I was playing with [hypothesis](https://github.com/HypothesisWorks/hypothesis-python) and basic ROS message types I found that out : 

```
alexv@AlexV-Linux:~/Projects/ros_pyros_ws$ source /opt/ros/indigo/setup.bash 
alexv@AlexV-Linux:~/Projects/ros_pyros_ws$ python
Python 2.7.6 (default, Oct 26 2016, 20:30:19) 
[GCC 4.8.4] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import rospy
>>> d=rospy.Duration(secs=0, nsecs=-1)
>>> d
rospy.Duration[-1]
>>> d.secs
-1
>>> d.nsecs
999999999
```

I am assuming this is bug ?

asmodehn, 2017-03-17:
or is it documentation ? 
The wiki says : duration secs/nsecs signed 32-bit ints, but genpy code has that : https://github.com/ros/genpy/blob/kinetic-devel/src/genpy/rostime.py#L41

I also found out that : 
```
>>> d.to_sec()
-9.999999717180685e-10
>>> d.to_nsec()
-1L
```
so the semantic is the same, but the data in slots needs to be interpreted before being usable.
Quite confusing...

So it might be fine after all, except that one cannot access the slots directly, and needs to know what method needs to be used to access the actual value...

closed 2017-03-17 by asmodehn

dirk-thomas, 2017-03-17:
A `Duration` is basically a different way to represent the float value of the duration in seconds. The seconds and nanoseconds are stored separately. In order to have a deterministic representation for every possible float value the `Duration` is normalized to contain a nanosecond part in the interval `[0, 999,999,999]`. Otherwise multiple duration instances with different values in the fields could represent the same float value.

It is basically the same as normalizing the angle in radians to an interval like `[0, 2 * pi)`.