ros/ros_comm #1021: rosmsg list not showing .msg files in subfolder

130s, 2017-03-18:
I reported at http://answers.ros.org/question/253386/ros-message-subfolder/ with reproducing steps.

dirk-thomas, 2017-03-18:
Please consider to provide a pull request to support this.

charmyoung, 2021-03-26:
@dirk-thomas 
This issue has influenced two command-line tools:
1. `rosmsg`/`rossrv` list not showing .msg files in the subfolder of ${package_name}/msg directory.
2. `rosmsg`/`rossrv` `show`/`md5` can not find msg information whose .msg  file is defined in the subfolder of ${package_name}/msg 
To solve this problems, three repos need to be fixed:
1. `rosmsg` package for `list` command and `show` command
2. `genmsg` package for `msg_loader` , given package and type
3.  `rosbash` package for Tab Completion

So I wonder if it is a mandatory requirements for the user to define the .msg or .srv file in the msg first-level directory only ?
If it is okay and allowed for user to define the .msg or .srv file in the subfolder of msg directory, I will provide the pull request for those repos mentioned above.

dirk-thomas, 2021-03-26:
@charmyoung I am not a maintainer of this repository anymore. Hopefully the new maintainer will follow up on this and answer your question.

charmyoung, 2021-03-27:
> @charmyoung I am not a maintainer of this repository anymore. Hopefully the new maintainer will follow up on this and answer your question.

Perhaps  @jacobperron would help?

jacobperron, 2021-04-05:
From a quick test, it seems like message generation works if a msg file is in a subdirectory of `msg/` (e.g. I can include and build against those messages), but the introspection tools (e.g. `rosmsg`) do not work.

@charmyoung I'm happy to review pull requests with a proposed fix.

closed 2025-05-31 by sloretz