ros/ros_comm #1022: [roslaunch] XML dump

ipa-mdl, 2017-03-20:
I have written a [little script](https://gist.github.com/ipa-mdl/66b14a895e13003a2f76e483bd00aa63) that dumps the fully parsed roslaunch XML. The output is sorted and can be launched right away.

We have used it for comparing changes while refactoring. In addtion it could be used to bundle launchfiles and config in a self-contained files. e.g. to decouple outdated versions.

Current limitations (@ipa-fxm: did I forget something?):
* `$(anon ..)` gets evaluated before the export
*  `clear_params` gets ignored

As other mights find this feature helpful as well, it would be great to include it with the roslaunch tool.
I guess it should be straight-forward to implement the XML output in the config object and add a CLI flag (`--dump-xml`?) to roslaunch.

Before putting more effort into it I'd like to have some feedback if this feature can be included and perhaps some implementation hints (single script vs. argument, which API not to touch etc.).

dirk-thomas, 2017-03-20:
Sounds like a helpful feature to me.

I am agnostic which of two options you would like to choose (separate script vs. argument). I think it mainly depends on if the new script needs some of the same command line arguments. If yes, adding it as an option avoids duplicating some of the command line argument logic I assume.

Anyway if it is a separate script you should move the logic into a `main` function and call it from within `if __name__ == '__main__':`.

mikepurvis, 2017-03-20:
Given the presence of `--dump-params` already, I'd vote for a flag.

closed 2025-05-31 by sloretz